Nishida-Lab/motoman_project #4: 一応動作確認しました

Ry0, 2015-11-02:
@RyodoTanaka 
なんかいろいろわかってないので**時間があれば**教えてください

##  wstoolの問題
READMEのとおりコマンドを打つと下のようなエラーが

```bash
Using initial elements from: src/motoman/.rosinstall
ERROR in config: Is not a local file, nor a valid URL [src/motoman/.rosinstall] : unknown url type: src/motoman/.rosinstall
```
？？？だけど
これが通らなくても対して問題なさそうなので手動で`motoman`のリポジトリをとってきて`indigo-devel`のブランチに切り替えました．

## cakin_make通らねえ問題
さっきのコマンドが通らなかったのが問題なのかわからないけど`industrial_msgs`がどうのとかいうので

```bash
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "industrial_msgs"
  with any of the following names:

    industrial_msgsConfig.cmake
    industrial_msgs-config.cmake

  Add the installation prefix of "industrial_msgs" to CMAKE_PREFIX_PATH or
  set "industrial_msgs_DIR" to a directory containing one of the above files.
  If "industrial_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  motoman/motoman_msgs/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ry0/workspace/ROS/motoman_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ry0/workspace/ROS/motoman_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
このリポジトリを取ってきました．

> ros-industrial/industrial_core
> ROS-Industrial core meta-package. http://ros.org/wiki/industrial_core
> https://github.com/ros-industrial/industrial_core

```bash
git clone https://github.com/ros-industrial/industrial_core.git
```
これでどうにか通りました．良道の環境にはもとから入っていたのかな．．．

##  確認
現段階ではこれでなんかいろいろやるんだよね？

```bash
roslaunch motoman_description sia5_rviz_demo.launch
```

なんかJoint State Publisherをうまいこと動かせるようになったら，おれもROSやってるっぽくなるのでできるだけがんばります

[image]

つくちゃ頑張って〜

RyodoTanaka, 2015-11-02:
@Ry0 
## 1. wstool について
すみません。READMEが間違ってました...orz
早急に修正します。
誤
```bash
wstool init src src/motoman/.rosinstall
```
正
```bash
wstool init src src/motoman_project/.rosinstall
```

## 2. catkin_make通らねぇ問題
上記のミスが起こると、`rosdep`しても必要なパッケージが入ってこないので、起きていたものと思われる。

## 3. Joint動かせねぇ問題
最新のデータをこれからUploadします。それは動きます。

以上、いろいろ迷惑かけてすみません。
よろしくお願いします。

RyodoTanaka, 2015-11-02:
@Ry0 
今、上記の修正を行いました。
試してみてください。
うまく行ったら、このisuueを閉じてください。
よろしくお願いします。

ちなみに、今のディレクトリでもう一回`wstool`しようとすると別のエラー（もう`.rosinstall`あるけど？的なエラー）がでると思うので、その場合は、`<catkin_ws>/src/.rosinstall`を削除してもう一度README通りにやってみてください。

Ry0, 2015-11-02:
Joint State Publisher動きました

```bash
rosdep install -i --from-paths src
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
motoman_driver: Cannot locate rosdep definition for [industrial_robot_client]
motoman_msgs: Cannot locate rosdep definition for [industrial_msgs]
motoman_sia20d_moveit_config: Cannot locate rosdep definition for [industrial_robot_simulator]
```
rosdepでしくじるのは相変わらずですが，他のリポジトリをクローンしたら解決できるので良しとします
ありがとう！

[image]

RyodoTanaka, 2015-11-12:
@Ry0 
私も、以下のエラーが起きました...
wikiを変更しておきます。

### 症状
`rosdep` しようとすると怒られる...orz
```bash
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
motoman_driver: Cannot locate rosdep definition for [industrial_robot_client]
motoman_msgs: Cannot locate rosdep definition for [industrial_msgs]
motoman_sia20d_moveit_config: Cannot locate rosdep definition for [industrial_robot_simulator]
```

もちろん、ガン無視して`catkin_make`しても、怒られる...
```bash
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "industrial_msgs"
  with any of the following names:

    industrial_msgsConfig.cmake
    industrial_msgs-config.cmake

  Add the installation prefix of "industrial_msgs" to CMAKE_PREFIX_PATH or
  set "industrial_msgs_DIR" to a directory containing one of the above files.
  If "industrial_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  motoman/motoman_msgs/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ryodo/Documents/ROS/motomam_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ryodo/Documents/ROS/motomam_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
### 解決方法
直接`apt-get`してくる
```bash
sudo apt-get install ros-indigo-industrial-msgs
sudo apt-get install ros-indigo-industrial-robot-client
sudo apt-get install ros-indigo-industrial-robot-simulator
```

Ry0, 2015-11-12:
ちょうどUbuntuをインストールし直しているので，こちらでも新しい環境でもう一度試してみます

RyodoTanaka, 2015-11-12:
@Ry0
了解です。
また結果を教えてくださいm(__)m

Ry0, 2015-11-12:
@RyodoTanaka 
全く新しい環境でも最低限の環境を構築できたことを確認しました．
これで他のひとも試すことができると思います．
ありがとう．

closed 2015-11-12 by Ry0