Nishida-Lab/motoman_project #18: Gazeboでうまく動かない問題　メモ

RyodoTanaka, 2015-11-28:
以下のは、`control`ブランチでの現状です。`master`ブランチにおける話ではありません。
Gazeboで動かそうと思って、デフォルトのコントローラである、JointStateControllerを用いている。
以下のコマンドでlaunchしてみたところ、最初だけlink_sが回転するが、その後にcommandを入力しても動かない。
```bash
roslaunch motoman_gazebo sia5_empty_world.launch
roslaunch motoman_control sia5_control.launch
```
rqtで見てみると、`/sia5/joint_s_position_controller/state/command`の値は動くが、`error`はリンクがどこにあっても`command`の値と一致してしまっている。またもちろん、`process_value`も3.14から動く気配がない。
rrbotの例と見比べて見ても何が原因で変な挙動を示すのかがさっぱりわからない。
因みに、rrbotの例を起動して、pidパラメータを変更してみると、pゲインが小さいと動きはするが、うまく目標値に追従できなくなることはわかった。
また、sia5に関してもpidパラメータをp=1.0にだけしており、これが原因かと思ったが、pゲインの値を変更すると、ros_control起動時のlink_sの回転速度が変化することから、とりあえずコントローラは動いているものと考えられる。
以上より、現状と問題点をわかる範囲でメモしておく。
### 現状
- gazeboがを立ち上げることができる。
- ros_controlerは最初にlink_sを回転させることはできる。

### 問題点
- ros_controlをlaunchした後、`/sia5/joint_s_position_controller/command/data`に値をpublishしてもGazeboのシミュレータが反応しない。
- rqtで`/sia5/joint_s_position_controller/state/error`を確認すると、目標値と現在地の差分を表せていないようである。

以上です。
何かわかる方がいらっしゃれば、是非教えてください。何卒よろしくお願いします。

mum254, 2015-11-28:
ぱっと見ました。これで動かないのは確かに深そうですね。どこに原因があるのかこの情報からではつかめませんね。gazeboのcontorollerの情報をもっと見られればいいのですが。
デバッグするにも、名前空間、7つのjointの名前・transmission・controllerが絡みあうため、容易ではなさそうですね。

原因の切り分けのため確認しますが、過去に一度でもgazebo上で動くことは確認しましたか？
なぜこれを聞くかというと、gazeboのチュートリアルと比較してxacroのマクロを活用している箇所が見られるからです。もし自分だったら、ひとまず下記のパターンで動きます。あくまで私の場合なので、参考までに。

・パターン1：過去にgazeboでmotomanを動かしたことがある。
　動いた時点までコミットをさかのぼり、やり直す。
　例えば、xacroのマクロ化する前は動いていたけど、マクロ化した結果動かなくなったということなら、単純な文法ミス。マクロ化する前までコミットをさかのぼり、一つ一つ動作を確実に確認しながら順番にマクロ化し直す。

・パターン2：実は、gazeboでmotomanを動かしたことがない。rrbotでしか動作を確認していない。
　最初からやりなおします。
　つまり、再度rrbotのサンプルを引っ張ってきて、それをベースにジョイントを一つ一つ確実に動作することを確認しながら、7つまで増やす。リンクパラメータをmotman仕様にしても動くかもこの段階で確認しておく。
joint名や、名前空間名を変えても動くかも、変数一つ一つレベルで確認し、ベースを作り上げる。ださい見てくれ以外は全部motomanのパラメータで動くことを保証しておきます。
　そして最後に、見てくれの部分を交換して、かっちょいいシミュレータに仕上げる。

もう少し軽かったら手を動かしてみようと思いましたが、後者だと当方手が回らなそうで。。。
クリティカルな意見が出せずすみません。これの成り行きは気になるので、また何かあったら挙げて下さい。

RyodoTanaka, 2015-11-28:
@mum254 
早速、ありがとうございます。
実は、パターン2ですので、最初からやり直してみたいと思います。
ちなみに現在は、リンクを1つ（link_s）だけにして試していたところです。
結果については、試し次第ご連絡いたします。
お忙しい中、本当にありがとうございました。

RyodoTanaka, 2015-11-28:
@mum254 
原因が判明しました！！！
[Github : RyodoTanaka/gazebo_ros_demos (branch : motoman_practice)](https://github.com/RyodoTanaka/gazebo_ros_demos/tree/motoman_practice)
上記のブランチでトライ＆エラーを行ったところ、`joint`の種類が原因であることがわかりました。
初期のrrbotでは、`joint`の種類に`continuous`を使用しているのですが、motomanでは、`revolute`を使用しています。理由は、実機の各`joint`に制限があるためです。
この`joint`の種類を使用するためには、`<limit>`要素と`<safety controller>`要素を追加する必要があります。この際に、`<safety controller>`要素にある、`k_position`と`k_velocity`を設定する必要が有ります。
この値がうまく設定できておらず、思った通りの動作ができていないことがわかりました。
正しい動作にするにはどのようにすれば良いかはまだわかっておりません。
以前`ros_control`を勉強した際にこれらについての説明も読みましたが、よくわからなくて適当な値を入れてしまっておりました。。。
どんな値を入れればよいのか再度勉強してきます。
いずれにせよ、原因は判明しました。アドバイス、本当にありがとうございました。これからもよろしくお願いします。

mum254, 2015-11-29:
@RyodoTanaka 
なるほど、理解しました。良かったですね。
こちらのリポジトリの`control`ブランチでは、ひとまず`continuous`にして動くようにしたということですね。

`revolute`の場合、`limit`は必須のようですが、`safety controller`はオプションのようですね。
http://wiki.ros.org/urdf/XML/joint

まずは`limit`だけ入れてみて動かし、機械的に制約がかかることを試すと良いかもしれませんね。

その後、`safety controller`でソフトウェア的な安全機構を入れて、限界値周辺で安全に停止することを確認すると良さそうですね。
http://wiki.ros.org/pr2_controller_manager/safety_limits
これ、説明が分かりにくいですね笑。解釈を書いておきますが、間違っていたらすみません。

とりあえず、`required`の`k_velocity`は試行錯誤でしょうね。
関節リミットにぶつけるのが怖ければ大きくする的な。

関節リミットにぶつけるのも確実に回避したいなら、`k_position`で速度とリミット値の関係に制約をかけなさいと言う風に見えます。
`safety_controller`が利き始めタイミングは`soft_hoge_limit`と`k_position`で調整できるっぽくて、`limit`で指定した限界値より前で停止するようにうまく設定しろと言うイメージですかね。
だから、`soft_lower_limit`は`lower`よりも大きく設定し、`soft_upper_limit`は`upper`よりも小さく設定するなんて書いてあるんですね。
よく分からなければ、ここは最悪全部デフォルトの0にすれば機能自体が無効になるのでしょう。

頑張ってください。

mum254, 2015-11-29:
あ、先ほどのサイトを読んで気付いたのですが、元々動かない理由が分かったかもしれません。
もともと、
```
<safety_controller k_position="0" k_velocity="0.5" soft_lower_limit="${s_axis_lower_limit}" soft_upper_limit="${s_axis_upper_limit}"/>
```
としてましたよね。`soft_lower_limit`と`soft_upper_limit`に値を設定しているのに、`k_position`を0にしたら、任意の角度で速度0を出し続けるので、動かないと思います。
　※先ほどのサイトのグラフの並行四辺形が、水平の直線になります。

動かしたいなら、`soft_lower_limit`と`soft_upper_limit`を0にすれば、速度抑制機能自体が停止して動き始めると推測しますが、いかがでしょうか？

RyodoTanaka, 2015-11-29:
@mum254 
ありがとうございます。
早速試してみます。
以前勉強した際に、
> safety_controllerが利き始めタイミングはsoft_hoge_limitとk_positionで調整できるっぽくて、limitで指定した限界値より前で停止するようにうまく設定しろと言うイメージですかね。
だから、soft_lower_limitはlowerよりも大きく設定し、soft_upper_limitはupperよりも小さく設定するなんて書いてあるんですね。

上記の部分のイメージがよくわかっておらず、wikiの図を見ながら、なんで傾きを設定するんだろう？何か基準は有るのだろうか？などと、とにかく何をしたいのかが把握できなかったため、今回のように、`k_position`と`k_velocity`だけでなく、`soft_lower_limit`と`soft_upper_limit`にも適当な値を入れてしまっておりました。。。

とりあえず、`safety_controller`を使うつもりは無いので、
> 動かしたいなら、soft_lower_limitとsoft_upper_limitを0にすれば、速度抑制機能自体が停止して動き始めると推測しますが、いかがでしょうか？

上記のようにアドバイス頂いた通りにやってみました。

しかし、うまくいかなかったので、試行錯誤した経緯を示します。

1.  `<safety_controller>`の全要素を0とする。
    -> 全関節が動かなくなる。

2. そもそも、 `<safety controller>`の部分をコメントアウトする。
    -> `link_s`だけが最初に半回転する。
    また、`joint_s`、`joint_r`、`joint_t`だけが動かなくなる。

以上が現状です。
1.については、正直よくわかりません。。。すみません。
2.については、モデルの記述が原因のように感じるので、見なおしてみようと思います。
よろしくお願いします。

RyodoTanaka, 2015-11-29:
@mum254 
モデルの原因わかりました。
`joint_s` `joint_r` `joint_t`だけ、`upper_limit = 180 [deg]` `lower_limit = -180 [deg]`であることが原因でした。
根本的解決に放っていませんが、それぞれのlimit角度を179.9 [deg] に設定しなおしたところ、うまく動くようになりました。
とりあえずcommitしますが、少し粘ってみます。

mum254, 2015-11-29:
@RyodoTanaka 
報告ありがとうございます。少しずつ見えてきていい感じですね。

><safety_controller>の全要素を0とする。
>-> 全関節が動かなくなる。

ちょっと不安でしたが、やはり0と明示的に設定するとダメみたいですね。Wikiのデフォルト0とかいう記述自体が怪しいですね。現状のように、コメントアウトするのが妥当ですね。

>upper_limit = 180 [deg] lower_limit = -180 [deg]

なるほど。この設定値だと、180度のところで連続と解釈され、理論的には`continuous`と等価になってしまっている可能性があります。しかし、これはあくまで仮定で、`controller`側がどう解釈するか分からないとなんともいえません。

少なくともこの設定が
>-> link_sだけが最初に半回転する。
>また、joint_s、joint_r、joint_tだけが動かなくなる。

という挙動を引き起こしていることは、切り分けできそうですね。

これは`controller`側の仕様またはバグの可能性がありますが、そのような記述は見つからないので、ソースコードまで覗かないと断定はできません。

実際のモデルも関節リミットが±180なのでしょうか？
そうであれば、暫定対策として現状の設定でもありかもしれませんね。

面白くなってきましたね。もし何かわかったら、また教えてください。

mum254, 2015-11-29:
`controller`云々言いましたが、モデル自体の話かもしれませんね。この辺りはどこがどう依存し合っているのか正確に理解していないので、`controller`の仕様かバグ、のとこは撤回しておきますね^^;
何か知見があれば是非教えて下さい。

RyodoTanaka, 2015-11-29:
@mum254 
ありがとうございます
現状では、前述の通り、関節角度を±180[deg]ではなく、±179.9999[deg]にしています。

先ほど、西田先生からLINEで教えていただいた内容ですと、特異点を回避するようにlimitを設定する必要があり、controllerの仕様というよりは、±180[deg]に設定すること自体、特異点を含んだ記述になってしまっているということだと思います。よく考えれば、その通りですね。。。もう少し気を配るべきでした。。。
ですので、もっと上手なlimit設定（モデルに合わせた設定値）が必要かとは思いますが、motomanの仕様には一応±180[deg]と記載してありますので、今のところは、現状通り±179.9999[deg]に設定しておきたいと思います。
9を小数点第何桁まで設定するかという問題がありますが、角度計算にはdouble型を使うことになるので、おそらく現状の設定で切り捨て誤差による特異点到達は無いかと思います。

以上、よろしくお願いします。

mum254, 2015-11-29:
@RyodoTanaka 
なるほど、特異点ですか。納得、したような感じです。
現象との関係をよく理解できていないので、今度議論させてください。

それから、特異点の定義によりますが、ピッタリではなく近傍でも十分暴走を誘発する可能性はあると考えます。

例えば、ロボットヤコビアンの特異点を考えてみます。
特異点ピッタリだとヤコビアンの逆行列の分母が0になり大暴走します。
特異点ぴったりでなくとも、近傍であれば、分母が0.0001とかになって、やっぱり暴走の危険性があります。

そういう意味での特異点でなければ現状の設定値でもいいと思いますが、本設定によりリミット周辺で暴走誘発の可能性があるなら、設定値を見なおしたほうが安全かもしれません。

という心配点を投下しておきます^^;

よろしくお願いします。

RyodoTanaka, 2015-11-29:
@mum254 
ご指摘ありがとうございます。
暴走を止めるという意味のlimitではありませんでしたが、今後の開発においてはもちろん考慮しなければならないと思います。
本当だったら、暴走に入らないようにする定義を<safety_control>で行う必要があるような気もします。
いずれにせよ、ロボット運動学の勉強が足りておりませんので、急いで勉強してきます。。。

mum254, 2015-11-29:
@RyodoTanaka 
矛盾した発言をして混乱を招いていて申し訳ないです。
つい、特異点、と聞いた瞬間ピクっと反応してしまいまして…。修士の時のトラウマがあります笑。

さて、整理した方がいいですね。

> 暴走を止めるという意味のlimitではありません
そうですね、同意します。
ここの`limit`はあくまで機構的なものを意味していて、暴走の回避はソフトウェアで実施すべきだと思います。それを簡易的に出来るようにするのが、`safety controller`ですね。
だから、ここで言う`limit`については、一旦暴走云々は忘れます。

となると、ここで言う「特異点」の定義は、いわゆるロボットの運動学における「特異点」とは異なる概念なのだろうかと思い、気になっています。ここが、先ほど「現象と結びつかない」と言った部分で、理解できていません。

これが現在のトピックですね。宜しくお願いします。

RyodoTanaka, 2015-11-29:
@mum254 
ありがとうございます。
私の不勉強から、誤解を招く表現を使ってしまい、申し訳ありません。
>ここで言う「特異点」の定義は、いわゆるロボットの運動学における「特異点」とは異なる概念なのだろうかと思い、気になっています

私は、「特異点」の定義が実はキチンとわかっておりません。
ゆえに、正しい定義での「特異点」として発言できているのかどうかもわかりません。。。
limitの値を±180[deg]にすると、upper_limitとlower_limitが一致してしまい、ある一点のjoint角度を定義すると、upperの方の角度なのかlowerの方の角度なのかわからなくなり（どちらともとることができるようになり）、それが所謂「特異点」なのだと考えておりました。なので、limit角度を±180[deg]に設定すると、シミュレータもどちらの角度なのかわからなくなってしまい、不可解な挙動を示しているのだと考えておりました。
以上が、私が考えていたことです。きちんと理解できているかどうかもわかっていないのに、用語だけ使用してしまい、ご迷惑をお掛けしました。申し訳ありませんでした。

勉強してきたいので、もしよろしければ、キーワードを教えていただけると大変助かります。
何卒よろしくお願いします。

mum254, 2015-11-30:
@RyodoTanaka 
そうそう、ほとんど同じ解釈です笑。
きっと±180で変なことになっちゃってんだろうなー、よくわからないけど、みたいな笑。

用語ですが、私はよくわかっていません笑。
ここで言う特異点とはそういうもの、ということでここは締めましょうね。今度face to face で議論させてください。

少なくとも知っているのは、ロボットの逆運動学における特異点と、ヤコビアンにおける特異点がそれぞれ存在すること、それぞれが特異点で取る姿勢は異なるということです。
前者は角度指令、後者は角速度指令なので、解釈が異なります。この辺りの話は、吉川先生のロボット制御基礎論という書籍でわかりやすく書いてあったと思います。ご参考までに。

こちらこそ今後とも色々教えて下さい。引き続きがんばってください。

RyodoTanaka, 2015-11-30:
@mum254
丁寧にお答えいただき、ありがとうございます！
ロボットの逆運動学における特異点（角度の特異点）とヤコビアンにおける特異点（角速度の特異点）の２種類があるということなんですね。勉強になります。
ちょうど今、慣性モーメントテンソルの計算に、概形図形を使って解こうとしていたので、ヤコビ行列、ヤコビアンは個人的にはとてもタイムリーな話題です！

また研究室でお会いする時にお話ができるよう、教えていただいた本、勉強しておきます。

closed 2016-02-10 by Ry0