Nishida-Lab/motoman_project #20: Could not load controller ???

Ry0, 2015-11-29:
@RyodoTanaka 

私の環境だけかもしれませんが
現時点でのmａｓｔｅｒの最新コミット
https://github.com/Nishida-Lab/motoman_project/commit/b88362197bc096b2ed16de82d22cfdc1f409cbb4
にて

まず最初のエラー
```bash
[ERROR] [1448809078.115189410]: SpawnModel: Failure - model name sia5 already exist.
```
と
```bash
[ERROR] [1448809081.753359785, 803.041000000]: Could not load controller 'joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.
[ERROR] [1448809081.753396377, 803.041000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
```
から
```bash
[ERROR] [1448809082.756997785, 804.040000000]: Could not load controller 'joint_s_position_controller' because controller type 'effort_controllers/JointPositionController' does not exist.
[ERROR] [1448809082.757036242, 804.040000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1448809083.758280] [805.039000] Failed to load joint_s_position_controller
```
のようなエラーが各リンクごとに発生します．

[image]

エラーの内容みてもさっぱりわからなかったので一応issueに残しておきます．．．

RyodoTanaka, 2015-11-29:
@Ry0 
まず、最初のエラーについて
```bash
[ERROR] [1448809078.115189410]: SpawnModel: Failure - model name sia5 already exist.
```
このエラーは実は私の環境でも出ます。
モデルを構築するときに名前空間を二重定義してしまっているようです。調査し、修正しておきます。
しかし、このエラーは、モデルが起動できない理由ではありません。

続いて。２つ目のエラーについて
エラーコードを見ていると、joint_state_controllerが読み込めていないようです。
joint_state_controllerは、各関節の制御に用いる角度情報を教えてくれるコントローラなので、これが起動できないとすべてのjointが起動できなくなってしまいます。
ゆえに、たくさんエラーが出てしまっているものと考えられます。
なぜ、joint_state_controllerが読み込めていないのかがわからないですが、パッケージが足りていないのかもしれません。
手間ですが、もう一度wiki記載の手順を行ってみてください（再インストールのひつようはありません）。
それでも起きるようであれば、おそらく`ros_control`パッケージがうまくインストールできていないことになるので、wikiを修正する必要が出てきます。
なかなか、他人の環境で試せないので、エラーばかり出てしまい申し訳ないのですが、ご協力よろしくお願いします。

Ry0, 2015-11-30:
```bash
model name sia5 already exist.
```
のエラーはたぶん最新のコミットで解決されてます．

次のCould not load controller関連のエラーは

```bash
sudo apt-get install ros-indigo-ros-controllers
```

とすることで解決しました．
ありがとうございます．

RyodoTanaka, 2015-11-30:
@Ry0 
以前は、`motoman_control`のpackage.xmlに`ros_controller`を記述していたのですが、catkin_makeの際に、Warningが出ていたため、なくしていました。
復活させておきます。

RyodoTanaka, 2015-11-30:
@Ry0 
今、復活させてみたところ、特にWarningが出ていませんでした。
代わりに、rosdepがうまくできなくなりました。

そういえば、前にこの箇所を消去した時には、コンパイル時のエラーではなく、rosdepできなくなるからでした。。。
すみません。きちんとissueを書くべきでした。（コミットログには書いてるかも。。。確認してませんし、確認する必要もなくなりましたが。。。）

というわけで、READEの方を修正しておきましたので、よろしくお願いします。
起動確認したら、連絡いただけるとうれしいです。

Ry0, 2015-11-30:
@RyodoTanaka 
自宅の環境でもgazeboの動作確認しました．
エラーも全く出てません．

[image]

moveitもあのとき作ったやつで今のところ動いてる感じです．
あとあと問題が起こりそうですが

[image]

RyodoTanaka, 2015-11-30:
@Ry0 
確認ありがとうございます！
Move it については、また問題が起きたらisuueしていきましょう。
とりあえずこの問題は片付いたので、closeします。ありがとうございました。

closed 2015-11-30 by RyodoTanaka