Nishida-Lab/motoman_project #21: Motoman やることメモ

RyodoTanaka, 2015-11-29:
### Gazebo モデルに関して
- [ ] inertia要素の改善
  massの再定義、慣性モーメントテンソルの計算（当面は近似物体で求める）

- [ ] transmission要素の改善
  各モータの減速比等を定義。hardware_interfaceについて掘り下げる必要あり。

### コントローラについて
現在は、デフォルトのPIDコントローラを用いている。また、制御方式が、位置制御（関節角度制御）である。  
Gazeboモデルがある程度固まってきたら、コントローラを設計する必要あり。
- [ ] 速度制御、力制御の実装

- [ ] 新たなコントローラの設計と実装
  手始めは、I-PDコントローラが良いと思われる。

### プランニングについて
Move it を用いたプランニングをとりあえずやってみる。
これには、すでにプランニングをやっている株丹君の力を借りたい。
自分たちで構築するプランニングがMove itを介したほうが良いのか、否か判定する。

- [ ] Move it で順・逆運動学を実装する。

### その他
- [ ] ロボット運動学の習得（順運動学、逆運動学）
- [ ] 剛体の力学再勉強（慣性モーメントを中心に）
- [ ] 座標変換の勉強
- [ ] 7軸ロボットの利点を見つける
  現状では、衝突空間の減少が主な利点である。西田先生のアドバイスによると、新たな７軸の強みを見つけると良い。つまり、衝突空間（位置）から一回微分した、速度における強み（タクト時間減少等）、二回微分した、力における強み（各関節の負担減少等）を見つけると良い。この他にも、グリッパ特有のプランニングを考案するというのも良い。例として、７軸＋多指ハンド等。
- [ ] 多指ハンドのデータダウンロードと仮制作

RyodoTanaka, 2015-11-29:
transmissionについてですが、減速比を大きく変えたりしてみましたが、特に挙動の変化が見られませんでした。
挙動を変化させるには、inertiaを変えるほうが良さそうです。Inertiaに、減速比を付加した値を記述する方が良いかもしれません。。。
というわけで、次はinertiaやっていきます。
hardware_interfaceについては、controllerを設計する際に深く関わりがあると思いますので、その際に検証したいと思います。

RyodoTanaka, 2016-02-10:
コントローラについてですが、先日 @mum254 さんと話をしていていくつか判明したことがあるので、備忘録と条法共有のために記しておきたいと思います。
## 現状
JoitPositionControlとJointTrajectoryControlを使っているが、
1. Kpゲインが2000とかで、デカすぎる気がする
2. Kdゲイン入れたら、入力値がすごくノイジーになる

っていうところから、なぜなのか調べてみました。

## 現時点でわかっていること
> そもそも、Gazeboから取って来ているPosition情報とVelocity情報は正しいのか？（どれくらいの精度があるのか？）
ODEから直接取ってきているみたい。ODE自体の精度については、[demura.net](http://demura.net/9ode/7876.html)にもあるとおり、大丈夫と思われる。

#### ros_controllersが何かおかしい
上記のコントローラでは、PID制御を用いているが、Velocity情報がないときのVelocityの値の算出方法が後退差分（１次精度）であり、PID制御を使う上では２次精度以上が求められるので、適切でないと思われる。
これが、Kdゲインを印加した時に現れる特異な値だと思われる。
またODEには、PID制御器で生成した制御入力の値が各関節のトルクの入力値として渡されているようである。つまり、モータとギアボックスモデルが制御対象としてモデル化されているのではなく、制御対象が**１**になってしまっていると考えられる。これがKpゲインが大きくなりすぎている（と感じる）原因と思われる。

## 今後どうするのか
今は実機があるため、発見された問題を本家リポジトリからcloneして修正後PRするのは、研究室のリソース的にも適切ではない。なので、こういった問題が存在しているということを共有することを現在はゴールとし、Gazeboのモデルはあくまでも、プランニングなどのソフトウェアが正常に動くかどうかの大雑把なテスト用として用いることにする。

mum254, 2016-02-14:
> ros_controlles が何かおかしい
について，一応フォロー．

`position_contoroller::VelocityJointController`,または`position_contoroller::EffortJointController`を使用する際，微分項の偏差を後退差分によって算出するため不適切じゃないか，のクダリです．

PIDコントローラでリアルタイムに速度を数値計算をすると，制御周期がROSの処理時間に依存するため，速度(⊿x/⊿t)の分母⊿tが不定となり，値がブレそうだということも，問題に上がると思います．

実は，`position_controller`を利用する際，**位置指令**だけではなく**速度制御**も与えることによって，その問題を回避できるパスが用意されています．その際，ODE(gazebo側)の現在速度を偏差計算に用いているはずです．
ただし，速度指令を経路情報から算出する場合は数値計算になるので，こちらの精度に対する問題は否定できませんが，少なくともPIDコントローラの中で差分計算をするよりはマシだと思います．計画経路からの差分計算では設計上都合のいい周期を設定できるので，どこかで値がふっ飛んだりすることは，通常のリーズナブルな経路では起こらない，と考えているからです．

精度の議論はさておきとしても，速度指令を追加で入力できる仕様について知りたい方は，こちらのまとめを参照ください．`ros_controls`のコードをトレースしてまとめました．
http://qiita.com/MoriKen/items/78b0ad8c1eae257646dd

制御は詳しくないので，ご意見あればぜひよろしくおねがいします．

RyodoTanaka, 2016-02-14:
@mum254 
`ros_controller`についてのフォローありがとうございます。
また，Qiitaのまとめもありがとうございます。とても勉強になります。
ここまで来たら、自分のコントローラを実装してみたいですね！
http://qiita.com/MoriKen/items/c29f653d03baffe5f0e2
をもう一度読んで起きます。

closed 2017-03-27 by NishidaTakeshiLab