Nishida-Lab/motoman_project #46: 実機を動かす時の仕様について

mum254, 2016-01-14:
標記についてちょっと教えてもらえますか？

gazeboを使った時のソフトウェア構成はなんとなく分かってきました．
実機の方がよく理解できていません．

ざっと見てsimとrealの違いが合ったのが，
`motoman_project/motoman_control/launch/sia5/` の中の`sia5_real_control.launch` と `sia5_sim_control.launch` かと思い中を見ました．

すると，`sia5_sim_control.launch`では`ros_control`の設定があるのですが，`sia5_real_control.launch`では`FollowJointTrajectory`型のActionサーバである`motoman_control_node`だけが起動されています．
実機では`JointTrajectoryController`を立ち上げる必要はないのでしょうか？

私の解釈では，下記のシーケンスでメッセージが送られていると思っていたので，controller_managerを起動することなく，`FollowJointTrajectory`だけを起動する`sia5_real_control.launch`では不十分なのでは？と思ったのですが，間違いでしょうか？

moveit: `move_group`
↓
action: `FollowJointTrajectory`
↓
controller_manager: `JointTrajectoryController`
↓
transmission: `EffortJointInterface`
↓
hardware_interface: `RobotHW`

おかしいところがあったらご指摘いただけますか？

また，
`motoman_sia5_moveit_config`内の`controllers.yaml`で`FollowJointTrajectory` を指定していると思うのですが，上記`sia5_real_control.launch`と両方設定がいるのでしょうか？
それとも`sia5_real_control.launch`はサンプル的な位置づけのファイルでしょうか？

RyodoTanaka, 2016-01-15:
@mum254 
わかりづらい構成になっており、申し訳ありません。
実機を動かす際の構成についてご説明したいと思います。
実機をMoveItで動かす際には、

1. `roslaunch motoman_control robot_interface_streaming_fs100.launch`
    - 実機が`JointTrajectory`を受け取り動作し、`JointState`を返すようにlaunch
    - xacroファイルを`robot_description`として読み取り、`robot_state_publisher`をlaunch

2. `roslaunch motoman_control robot_real_control.launch`
    - `FollowJointTrajectory`を受け取り、`JointTrajectory`をpublishするようにlaunch

3. `roslaunch motoman_sia5_moveit_config moveit_planning_execution.launch`
    - 起動計画を`FollowJointTrajectory`でpublishするようにlaunch
    - MoveitのGUI(Rviz)をlaunch

の順でlaunchする必要があります。
メッセージの流れとしては、以下の通りです。

Moveit
↓
↓ FollowJointTrajectory　(action_libメッセージ)
↓
motoman_control_node
↓
↓ JointTrajectory (Topicメッセージ)
↓
motoman_driver
↓
↓ ros_industrialが定めたMotoPlusとの通信プロトコル
↓
実機

で、ここからが少しわかりづらいのですが、Moveitはロボットをどんなコントローラで動かしているのかを知るために、`<motoman_sia5_moveit_config>/config/controllers.yaml`を読んでいます。
なので、実機を動かしたい際には、ここに記述しているコントロールTopicを受け取って動かすようなノードを作成する必要があり、本リポジトリではそれが`motoman_control_node`になっています。また、この`motoman_control_node`を起動するために`motoman_control sia5_real_control.launch`をlaunchしています。

一方、Gazeboで起動する際には、そもそもGazeboの中で各関節がどんなコントローラを使用するのか記述し、そのコントローラを記述しておく必要があります。本リポジトリでは`<motoman_control>/config/sia5/sia5_joint_trajectory_control.yaml`にコントローラの記述を行い、`motoman_control sia5_sim_control.launch`でこれを読み込み、実行しています。Gazeboで起動する際には、これだけで`FollowJointTrajectory`メッセージを受け取って動けるようになっています。

以上より、`sia5_real_control.launch`と`sia5_sim_control.launch`は名前は似ていますが、中でやっていることは全然ちがうということです。わかりづらくて申し訳ありません。

よろしくお願いします。

mum254, 2016-01-15:
@RyodoTanaka 
詳しく教えてくれて，ありがとう．理解しました．

実機とgazeboで共通なのは`FollowJointTrajectory`のactionメッセージまでですね．

下記サイトの図が脳裏にあったので，Hardware Resource Interface Layerである`EffortJointInterface`までが共通で，`hardware_interface`の`RobotHW`と`RobotHWSim`で違いがあるのかと思ってしまっていました．
http://gazebosim.org/tutorials?tut=ros_control

`motoman_driver`を使う場合は，上記サイトの`Controller Manager`以降のインターフェースを吸収しているようですね．だから，現在のリポジトリにソフトウェア構成がそうなっているのだと理解しました．
確かに，`motoman_driver`がこの仕様となると，`sia5_real_control.launch`と`sia5_sim_control.launch`の中身が全く異なる仕様となるのは，当然の帰着ですね．こちらの方こそ勉強不足でして，教えてもらってスッキリしました．

ここまで踏まえて一旦知識を整理したいので，直接話をさせてもらえませんか？
1/19のAMは居室にいるのですが，都合如何でしょうか？

RyodoTanaka, 2016-01-15:
@mum254 
確かに、
http://gazebosim.org/tutorials?tut=ros_control
にある図ですと、森田さんのおっしゃる通りだと思います。
motomanがやってきた際に安川の方が説明されていたのですが、MotoPlusには実機のmotomanには到達点とそこまでの経過時間を指定すると、速度などを補完して動作するプログラムが起動しているので、ROSからは`JointTrajectory`メッセージを送れば良いようになっているようです。つまり、MotoPlusが`Controller Manager`以降のインタフェースを吸収していることになるのではないかと思います。
そういえばmotomanだけに限った話ではないのですが、下記サイトのスライドによると`ros_control`による制御はあまり精度が高くないようです。(全く検証していないので、どれくらいの精度の話なのか、そもそも事実なのかどうかもよくわかりませんが。。。)
http://www.slideshare.net/akio19937/1rosrosgazebo
提示いただいたGazeboと実機の概念図だと、内部の制御も同じになるように設計されているようですが、上記の話が本当なら、幾分か困難な部分もあるようです。。。
いずれにせよ、この点については不確かなことばかりです。思い出したので自分のためにも一応書いておきました。

1/19のAMでしたらおります。ちなみに何時ごろお越しの予定ですか？
もしよろしければお知らせいただけると幸いです。

よろしくお願いします。

mum254, 2016-01-15:
@RyodoTanaka 
なるほど，`ros_control` はまだ発展途上と言ったところですか．余談ですが，HSRがどうなっていたのか気になりますね．

1/19はAM07:00から在室する予定です．よろしくおねがいします．

mum254, 2016-01-18:
1/19(火)は強風が予想されています．電車が遅れる可能性があるので，到着が遅くなるかもしれません．
一応着いたら一言書きます．

RyodoTanaka, 2016-01-18:
@mum254 
承知しました。
気をつけてお越しください！

mum254, 2016-01-18:
すみません，当日判断となり恐縮ですが，本日は登校を控えます．

今朝，想定よりも雪があり，この先の電車の運行状況に影響する可能性があります．公共交通機関が止まり，移動できなくなる可能性を避けるため，朝から出社に切り替えます．ドタキャンになり申し訳ありません．

来週改めて伺います．それまで当方でも更に情報を整理しておきますので，よろしくおねがいします．

RyodoTanaka, 2016-01-18:
@mum254 
承知しました。
そちらは雪が積もっているんですね。。。
戸畑は特に積もっておりませんのでびっくりです！
また来週お会いできるのを楽しみにしています。

RyodoTanaka, 2016-01-20:
@mum254 
本日のPullRequestで実装したGazeboのモデルは、改善により、かなり実機に近い動きをするようになりました。
最新のmasterブランチにて実装してありますので、よろしければお試しください。
なお、最新のモデルには、
- sia5
- sia5を載せる台
- ジャミンググリッパ
- KinectV2

が実装されています。
これらのモデルを用いたMoveitによるモーションプランニングも行えます。
具体的には、

1. Gazeboモデルの立ち上げ
    ```bash
    roslaunch motoman_gazebo sia5_nishida_lab.launch
    ```

2. Moveitの立ち上げ
    ```bash
    roslaunch motoman_sia5_moveit_config moveit_plannning_execution.launch
    ```
    ただし、Rviz上のMoveitプラグインが一発では表示されないかもしれないので、その場合は、
    `<motoman_project>/motoman_sia5J_moveit_config/launch/moveit.rviz`を選択してください。

以上の手順で行えます。

今回行ったモデルへの変更は具体的には、モデル自身にかかる重力をOFFにしたという点です。
これは、`<motoman_project>/motoman_description/urdf/sia5/body.gazebo.xacro`に`<turnGravityOff>true</turnGravityOff>`を追加することで実現しました。
また、この手法は、PR2でも行われているようです。（具体的には、PR2の手先部分でこれを取り入れている）

以上、よろしくお願いします。

P.S
シミュレーションには結構なマシンパワーを必要とします...なので、もし試される場合にはそれ相応のマシンを用意していただくか、ROSの得意分野である、リソースの分散によるシミュレートをおすすめします。

mum254, 2016-01-21:
@RyodoTanaka 
ありがとうございます．後ほど見てみます．

mum254, 2016-01-24:
@RyodoTanaka 
次，1/29(金)AMから14:00位まで在室予定です．色々教えて下さい．
グリッパやKinect付きのモデル，興味あるので，登校までにできる範囲でチェックします．

あと，先日登校できなかった分の穴埋めと言ってはなんですが，`Controller`と`HardwareInterface`周りのフローに関する情報をまとめてみました．よろしければ，チェックください．
http://qiita.com/MoriKen/items/613635b90f3a98042dc5
ようやっとこの辺りの仕組みが分かってきました．．．

RyodoTanaka, 2016-01-24:
@mum254 
1/29(金)の件、承知しました。
こちらこそよろしくお願いします。

また、作成頂いたサイト、本当にありがとうございます。
なんとなくしか理解できていなかった部分等、とても勉強になります！
多分、日本語でここまで詳しく`Gazebo`と`ros_controller`の周辺を取り扱っているサイトは無いと思います。
私が申し上げるのもおこがましいですが、本当に素晴らしいと思います！
宣伝して回りたいです！というか、私の知り合いに宣伝して回ってもよろしいですか?
いずれにせよ、英語の本にはここまで詳細な情報があるという事実に、英語学習の大切さを痛感しました。。。orz
とにかく、ありがとうございました。
1/29にお会いできるのを楽しみにしております。

mum254, 2016-01-25:
@RyodoTanaka 
そうそう，この辺の大した情報が見つからなくて，自分で書いちゃえって感じでした．

実は，紹介している英語本の方の説明も大したことは書かれていなくて笑，あーもー！ってなって，自分で調べて書いています．一応筋が通る様に紐解いたつもりですが，私の解釈が誤っている可能性はあります^^;

ですので，宣伝していただけるとのこと，歓迎です！！ぜひ，ROS使いの先輩の皆さんに見て頂き，誤りや気づいたことがあれば指摘をして欲しいくらいです．

いずれにしろ，現状のROS本に関しては，和書より洋書の方が圧倒的に充実していると言えます．何冊か読みましたが，洋書の方が遥かに広い内容を扱っており，解説も詳しいです．入門本はどんどん改版されており，最近はかなりアドバンスドな内容を扱う書籍も出てきています．図書館に置いてある本もあるようですので，ぜひお手に取ってみると良いと思います^^

和書については，先生の本がブレイクスルーになると思っています！

それでは，また後日よろしくお願いします．

closed 2016-06-23 by RyodoTanaka