Nishida-Lab/motoman_project #50: Gazeboの実行について

taka1103xyz, 2016-01-24:
wiki通りにやって見たのですが、下記エラーが起きました。
```bash
started roslaunch server http://fujimoto-All-Series:47464/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /sia5/joint_state_controller/publish_rate: 50
 * /sia5/joint_state_controller/type: joint_state_contr...
 * /sia5/sia5_controller/gains/joint_b/d: 0.0
 * /sia5/sia5_controller/gains/joint_b/i: 1.0
 * /sia5/sia5_controller/gains/joint_b/i_clamp: 100.0
 * /sia5/sia5_controller/gains/joint_b/p: 10.0
 * /sia5/sia5_controller/gains/joint_e/d: 0.0
 * /sia5/sia5_controller/gains/joint_e/i: 10.0
 * /sia5/sia5_controller/gains/joint_e/i_clamp: 100.0
 * /sia5/sia5_controller/gains/joint_e/p: 100.0
 * /sia5/sia5_controller/gains/joint_l/d: 0.0
 * /sia5/sia5_controller/gains/joint_l/i: 500.0
 * /sia5/sia5_controller/gains/joint_l/i_clamp: 500.0
 * /sia5/sia5_controller/gains/joint_l/p: 1000.0
 * /sia5/sia5_controller/gains/joint_r/d: 0.0
 * /sia5/sia5_controller/gains/joint_r/i: 1.0
 * /sia5/sia5_controller/gains/joint_r/i_clamp: 100.0
 * /sia5/sia5_controller/gains/joint_r/p: 10.0
 * /sia5/sia5_controller/gains/joint_s/d: 0.0
 * /sia5/sia5_controller/gains/joint_s/i: 10.0
 * /sia5/sia5_controller/gains/joint_s/i_clamp: 100.0
 * /sia5/sia5_controller/gains/joint_s/p: 100.0
 * /sia5/sia5_controller/gains/joint_t/d: 0.0
 * /sia5/sia5_controller/gains/joint_t/i: 1.0
 * /sia5/sia5_controller/gains/joint_t/i_clamp: 100.0
 * /sia5/sia5_controller/gains/joint_t/p: 10.0
 * /sia5/sia5_controller/gains/joint_u/d: 0.0
 * /sia5/sia5_controller/gains/joint_u/i: 10.0
 * /sia5/sia5_controller/gains/joint_u/i_clamp: 100.0
 * /sia5/sia5_controller/gains/joint_u/p: 100.0
 * /sia5/sia5_controller/joints: ['joint_s', 'join...
 * /sia5/sia5_controller/type: effort_controller...
 * /use_sim_time: True

NODES
  /sia5/
    controller_spawner (controller_manager/spawner)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[gazebo-1]: started with pid [3007]
process[gazebo_gui-2]: started with pid [3011]
process[urdf_spawner-3]: started with pid [3015]
process[sia5/controller_spawner-4]: started with pid [3016]
process[robot_state_publisher-5]: started with pid [3017]
Gazebo multi-robot simulator, version 2.2.3
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 2.2.3
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Msg Waiting for master.[ INFO] [1453668633.997044852]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1453668633.997164229]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Msg Waiting for master
Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 192.168.0.18
spawn_model script started
[INFO] [WallTime: 1453668634.089112] [0.000000] Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [WallTime: 1453668634.184016] [0.000000] Loading model xml from ros parameter
[INFO] [WallTime: 1453668634.186240] [0.000000] Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1453668634.237436381, 1584.218000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1453668634.272797749, 1584.253000000]: Physics dynamic reconfigure ready.
[INFO] [WallTime: 1453668634.491076] [1584.465000] Calling service /gazebo/spawn_urdf_model

Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 192.168.0.18
[ INFO] [1453668635.002197807, 1584.761000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[INFO] [WallTime: 1453668635.003451] [1584.761000] Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1453668635.030079473, 1584.761000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[urdf_spawner-3] process has finished cleanly
log file: /home/fujimoto/.ros/log/161039cc-c2ca-11e5-be16-0862664d096c/urdf_spawner-3*.log
[ INFO] [1453668641.130464698, 1584.761000000]: Loading gazebo_ros_control plugin
[ INFO] [1453668641.130577640, 1584.761000000]: Starting gazebo_ros_control plugin in namespace: /sia5
[ INFO] [1453668641.131211180, 1584.761000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ INFO] [1453668641.319304747, 1584.761000000]: Loaded gazebo_ros_control.
[INFO] [WallTime: 1453668641.323952] [1584.765000] Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [WallTime: 1453668641.325897] [1584.768000] Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [WallTime: 1453668641.327296] [1584.769000] Loading controller: joint_state_controller
[ERROR] [1453668641.329509091, 1584.772000000]: Could not load controller 'joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.
[ERROR] [1453668641.329554039, 1584.772000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1453668642.330885] [1585.770000] Failed to load joint_state_controller
[INFO] [WallTime: 1453668642.331268] [1585.770000] Loading controller: sia5_controller
[ERROR] [1453668642.333144621, 1585.772000000]: Could not load controller 'sia5_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist.
[ERROR] [1453668642.333192662, 1585.772000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1453668643.333990] [1586.764000] Failed to load sia5_controller
[INFO] [WallTime: 1453668643.334572] [1586.764000] Controller Spawner: Loaded controllers: 
[INFO] [WallTime: 1453668643.337620] [1586.768000] Started controllers: 
^C[robot_state_publisher-5] killing on exit
[sia5/controller_spawner-4] killing on exit
[INFO] [WallTime: 1453668661.727264] [1605.111000] Shutting down spawner. Stopping and unloading controllers...
[gazebo_gui-2] killing on exit
[gazebo-1] killing on exit
^C^C^C^C[sia5/controller_spawner-4] escalating to SIGTERM
[sia5/controller_spawner-4] escalating to SIGKILL
Shutdown errors:
 * process[sia5/controller_spawner-4, pid 3016]: required SIGKILL. May still be running.
shutting down processing monitor...
... shutting down processing monitor complete
done
```
よろしくお願いします。

RyodoTanaka, 2016-01-24:
@taka1103xyz 
エラー承知しました。
おそらく、`ros_controllers`の依存関係が満たせていない（rospackageがインストールされていない)ために起きたためと思われます。。。
とりあえず、
```bash
sudo apt-get install ros-indigo-ros-controllers
```
を実行した後にもう一度試してみてください。

いずれにせよ、`ros_controllers`の依存関係を`package.xml`に追加しておきます。。。

taka1103xyz, 2016-01-24:
@RyodoTanaka 
解決しました。
ありがとうございました！

解決したので、閉じます。

closed 2016-01-24 by taka1103xyz