Nishida-Lab/motoman_project #55: motomanでハンドリングを行いたい

Ry0, 2016-02-10:
タイトルの通り，このmotoman_projectでもmoveitを使ったPick and Placeを行いたいと考えております．
そこで「**Learning ROS for Robotics Programming - Second Edition**」，「**Mastering ROS for Robotics Programming**」の共に１０章にある指南にしたがって，graspを試みました．

それぞれの教科書が提供しているサンプルリポジトリは以下の通りです．

>Learning_ROS_for_Robotics_Programming_2nd_edition
>https://github.com/AaronMR/Learning_ROS_for_Robotics_Programming_2nd_edition  
>mastering_ros
>https://github.com/qboticslabs/mastering_ros

これらのサンプルが共通に使っているパッケージである「[moveit_simple_grasps](https://github.com/davetcoleman/moveit_simple_grasps)」があります．このパッケージが実にやっかいで，本家リポジトリでもサポートは終了しており，教科書に書いてあることが再現できておりません．（現段階ではmotomanは持ち込まずに，それぞれの教科書が提供してくれているロボットで試しています．）
ここからはそれぞれの教科書別に抱えている問題について書いていこうと思います．

## Learning ROS for Robotics Programming - Second Editionの場合
moveit_simple_graspsは`sudo apt-get install ros-indigo-moveit-full`をした拍子にインストールされたものを使用しています．
そもそもmoveit_simple_graspsはbaxter等の双腕を想定したプログラムの構成になっています．
したがって下のlaunchファイルをlaunchすると以下のようなエラーが
```bash
roslaunch rosbook_arm_pick_and_place grasp_generator_server.launch
```

```bash
process[moveit_simple_grasps_server-1]: started with pid [22820]
[ERROR] [1455088144.657619214]: Grasp configuration parameter `pregrasp_time_from_start` missing from rosparam server. Did you load your end effector's configuration yaml file? Searching in namespace: /moveit_simple_grasps_server/right
```

## Mastering ROS for Robotics Programmingの場合
この事態を把握しているようで，このセクションのプログラムを実行する場合，同梱しているものを使ってください．との記述がありますが，これもうまくいきませんでした．エラー内容はcloseされているissueの内容と全く一緒でした．

> 15日前にクローズされたもの
> Moveit_simple_grasping#3 
> https://github.com/qboticslabs/mastering_ros/issues/3

```bash
roslaunch seven_dof_arm_gazebo grasp_generator_server.launch
```

```bash
process[moveit_simple_grasps_server-1]: started with pid [1165]
moveit_simple_grasps_server: /usr/include/boost/smart_ptr/shared_ptr.hpp:653: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const moveit::core::RobotModel; typename boost::detail::sp_member_access<T>::type = const moveit::core::RobotModel*]: Assertion `px != 0' failed.
```

## こまった
これができるようになると，motomanでできることの幅が広がるので，是非とも実現したいのですが私の実力じゃこんな感じでした．
もし解決策がありましたらよろしくお願いします．

RyodoTanaka, 2016-02-10:
@Ry0 
根本的な解決とは程遠いですが、jammingもしくは、universalグリッパを使うなら、手先を指定の位置に持って行けばOKなので、研究室にあるハンドなら、とりあえずのハンドリングは実現できると思います。

もちろん、これではダメなのですが、そもそもハンドリングをやらせてみるという目的は一旦解決できるかと思うので、これをまずは試してみたいと思います。現時点で、手先位置・姿勢は自由に動かせるようになっているので、実現に必要なコストを考慮しても、とりあえずの解決策としては有効（僕たちがやりたいことだけですけど。。。）だと思います。

根本的な解決には時間が必要ですね。。。orz

mum254, 2016-02-11:
@Ry0 
「**Learning ROS for Robotics Programming - Second Edition**」の方で、`simple_grasps_server`を動かす方法が分かりました。

↓こんな感じに動いた
[image]
[image]

## 原因
肝心の[こちらのプルリク](https://github.com/davetcoleman/moveit_simple_grasps/pull/16)のマージ先が、`davetcoleman:hydro-devel `になっていて、`indigo`に対応していないっぽい。

## とりあえずやったこと
じゃあ、`moveit_simple_grasps`だけ`hydro-devel`にして`catkin_make`して動けばいいな、と思ったら案の定コンパイルエラー。

## 解決策
ということで、[こちらのプルリク](https://github.com/davetcoleman/moveit_simple_grasps/pull/16)と同じ変更を`indigo-devel`に施して`catkin_make`したら、動きました。手順は、テキスト通りでいいです。

## 変更後のファイル
このリポジトリにプルリクしておきます。動作確認したら、消してしまって構いません。

mum254, 2016-02-11:
あ、プルリクまでしなくていいですね。

`fix-moveit_simple_grasps`というブランチを作成し、`fix_moveit_simple_grasps`というフォルダにファイルを格納しておきました。中のファイルを、`moveit_simple_grasps`パッケージの所定の場所に上書きして、`catkin_make`しなおして下さい。

# コピー先
`fix_moveit_simple_grasps/simple_grasps_server.cpp`
→ `moveit_simple_grasps/src/imple_grasps_server.cpp` へコピー上書き

`fix_moveit_simple_grasps/grasp_generator_server.launch` (こっちはコピーしなくても`rosbook_arm`の動作には関係ありませんが、一応)
→ `moveit_simple_grasps/launch/grasp_generator_server.launch` へコピー上書き

RyodoTanaka, 2016-02-11:
@mum254 
ありがとうございます。
早速試そうとしたのですが、コンパイルエラーが出てしまったので、最新の`fix-moveit_simple_grasps`ブランチにて修正しました。
今から`pick`を試してみます。

RyodoTanaka, 2016-02-11:
@mum254 
@Ry0 
試したところ、Mastering ROS for Robotics Programmingと全く同じエラーが出てしまい、動きませんでした。。。orz
```bash
process[moveit_simple_grasps_server-1]: started with pid [19856]
moveit_simple_grasps_server: /usr/include/boost/smart_ptr/shared_ptr.hpp:653: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const moveit::core::RobotModel; typename boost::detail::sp_member_access<T>::type = const moveit::core::RobotModel*]: Assertion `px != 0' failed.
```
これがエラーです。と言っても、以前のものと一緒なのですが。。。

Ry0, 2016-02-11:
@mum254
@RyodoTanaka 
ご連絡が遅れました．
私の環境でも，同じエラーが出て困っています．

私もmoveit_simple_graspsのissueなどいろいろ辿って自分なりにコードを書き換えたのですが，同じエラーで詰まっています．
apt経由で導入した`moveit_simple_grasps`ものは削除して試しています．

mum254, 2016-02-11:
困りましたね。なぜパソコンに依存するのか…

取り急ぎ問題提起だけ。

ざっとしか見ていませんが、そのエラーは、shared pointerの初期化が失敗した時にでるっぽいですね。
http://stackoverflow.com/questions/3541179/shared-ptr-assertion-px-0-failed

メモリの競合が発生する要因等は無いでしょうか？

RyodoTanaka, 2016-02-12:
@mum254 
@Ry0 
メモリの競合ということで、試しに
`roslaunch rosbook_arm_pick_and_place grasp_generator_server.launch`
だけ実行して見たのですが、同様のエラーが出ました。なので、他のパッケージとの競合では無いと思います。
ただ、一点だけ変わったのは、以前であればエラーが表示されたあと自動でプログラムが終了されていたのですが、今回は自動でプログラムが落ちることはなかったです。とはいえ、これは本質的にはなんの関係もなさそうです。。。
あまり有益な情報ではありませんが、一応報告しておきます。よろしくお願いします。

RyodoTanaka, 2016-02-12:
@Ry0 
@mum254 
# 動きました！！！！！
[image]
### やったこと
`visual_tools_->loadEEMarker(grasp_data_.ee_group_);`
をコメントアウトする。

### 経緯
どこからプログラムが落ちているのか調べていたら、
`visual_tools_->loadEEMarker(grasp_data_.ee_group_);`
の部分でした。
この関数の作者は、`moveit_simple_grasp`の製作者と同じ方のようです。
この関数は本家リポジトリのPRを引っ張ってきている部分ですが、コンパイルすると
```bash
/home/ryodo/Documents/Lab/ROS/study_ws/src/moveit_simple_grasps/src/simple_grasps_server.cpp: In constructor ‘moveit_simple_grasps::GraspGeneratorServer::GraspGeneratorServer(const string&)’:
/home/ryodo/Documents/Lab/ROS/study_ws/src/moveit_simple_grasps/src/simple_grasps_server.cpp:154:53: warning: ‘bool moveit_visual_tools::MoveItVisualTools::loadEEMarker(const string&)’ is deprecated (declared at /opt/ros/indigo/include/moveit_visual_tools/moveit_visual_tools.h:169) [-Wdeprecated-declarations]
    visual_tools_->loadEEMarker(grasp_data_.ee_group_);
```
のようなワーニングが出ていました。
調べてみると、`loadEEMarker(const std::string&)`という関数ですが、代入するときに`const`ではないため、ワーニングが起きているようです。これが原因ではないと思うのですが、いずれにせよ、この関数から動かなくなっているというのが原因でした。
この関数がやっているのは、エンドエフェクタの名前を渡すだけす。また、クラス全体はmoveitのvisualに関することをやっていると判断しました。
なので、腕を動かすには本質的にはいらないだろうと判断し、コメントアウトしました。

## 今後どうするか
原因箇所の切り分けととりあえずの改善、動作確認は出来ましたが、実際には完全な修正ができていません。
また、何よりよくわからないのが、エラーが起きる原因はパソコンの環境にも依存していそうであるということです。このあたりは後々に大きなバグになって襲ってきそうなので、継続的に原因究明をしていきたいです。

以上、よろしくお願いします。

mum254, 2016-02-12:
@Ry0 @RyodoTanaka 
すばらしい！落ちた関数をどうやって調べたのか、あとで教えて下さい！

## `loadEEMarker`で落ちる箇所の推察
さて、気になる`visual_tools_->loadEEMarker(grasp_data_.ee_group_);`に話がシフトしますね。

もともとのエラーメッセージで、`const moveit::core::RobotModel;` という文字列があったことから、この型を扱う辺りが怪しそうな気がします。

`MoveItVisualTools::loadEEMarker()`の定義を見てみると、`RobotModel`のインスタンスを扱っているステップがありますね。
https://github.com/davetcoleman/moveit_visual_tools/blob/jade-devel/include/moveit_visual_tools/moveit_visual_tools.h#l171
```
RVIZ_VISUAL_TOOLS_DEPRECATED
  bool loadEEMarker(const std::string &ee_group_name)
  {
    return loadEEMarker(robot_model_->getJointModelGroup(ee_group_name));
  }
```
実装の中まで入っているとしたら、下記の箇所。
https://github.com/davetcoleman/moveit_visual_tools/blob/jade-devel/src/moveit_visual_tools.cpp#l251
```cpp
const moveit::core::LinkModel* ee_parent_link = robot_model_->getLinkModel(ee_parent_link_name);
```

## ここまでの現象とエラーメッセージからの原因の推察
- `robot_model_`インスタンスの居所は共有メモリなのでしょうが、うまく初期化がされていないために、NULL ポインタを指してしまって、`Assertion `px != 0' failed.`というエラーで落ちたのではないかと推察されます。

もし、このポインタが問題であるとしたら、`visual_tools_`のインスタンス生成時の、引数付きコンストラクタで初期化が失敗していることとなります。
```cpp
MoveItVisualTools::MoveItVisualTools(const std::string& base_frame, const std::string& marker_topic,
                                     robot_model::RobotModelConstPtr robot_model)
  : RvizVisualTools::RvizVisualTools(base_frame, marker_topic)
  , robot_model_(robot_model)
  , mannual_trigger_update_(false)
{
}
```

すると、気になるのはここです。
https://github.com/davetcoleman/moveit_simple_grasps/blob/indigo-devel/src/simple_grasps_server.cpp#l146
```
      // ---------------------------------------------------------------------------------------------
      // Load the Robot Viz Tools for publishing to Rviz
      visual_tools_.reset(new moveit_visual_tools::MoveItVisualTools(grasp_data_.base_link_));
      visual_tools_->setLifetime(120.0);
      visual_tools_->loadEEMarker(grasp_data_.ee_group_);
```
`loadEEMarker`を呼ぶ直前に、`reset()`メソッド呼んでいます。`new moveit_visual_tools::MoveItVisualTools(grasp_data_.base_link_)`で生成したインスタンスで上書きしていると推測されますが、引数に`robot_model`型のインスタンスを渡していないので、`loadEEMarker`使用時に`robot_model_`がヌルポになって落ちてしまうのは、分かるような気がします。

ですが、じゃあなぜ私の環境では動いているのか、説明ができなくなってしまいます。もう少し深堀しなければなりません。

てか、ここまで深堀しなければならないような状態に陥っていること自体、ライブラリとして解決すべき問題なのかもしれませんね…。

## 気になること
`loadEEMarker`の定義に、`RVIZ_VISUAL_TOOLS_DEPRECATED`と書いてあるのが気になります。もう推奨されていないってことでしょうか？

mum254, 2016-02-12:
環境に依存する原因が、当方のパッケージのバージョンが古いためでした。

下記のコミットのtestコード側も見ると、クライアント側のコードも少し変わっているようなので、これが原因だと思います。
新しいクライアントの方では、robot_modelに関する何かしらのグループのポインタを取得して、visual_tools_に渡す処理が追加されています。
src/grasp_filter_test.cpp
```
// 特に臭うところ抜粋。
-      visual_tools_->publishAnimatedGrasps(possible_grasps, grasp_data_.ee_parent_link_);      
 -      visual_tools_->publishIKSolutions(ik_solutions, planning_group_name_, 0.25);
 +      const robot_model::JointModelGroup* ee_jmg = planning_scene_monitor_->getRobotModel()->getJointModelGroup(grasp_data_.ee_group_);
 +      visual_tools_->publishAnimatedGrasps(possible_grasps, ee_jmg);
 +      const robot_model::JointModelGroup* arm_jmg = planning_scene_monitor_->getRobotModel()->getJointModelGroup(planning_group_name_);
 +      visual_tools_->publishIKSolutions(ik_solutions, arm_jmg, 0.25);
```
https://github.com/davetcoleman/moveit_simple_grasps/commit/e2e01981fe106b2899963177d9756fc988cdb3a1

これなら、前述した`RobotModel`型のポインタがヌルポになる理由も説明がつきそうです。
pythonからどうインターフェースするかまでは分かりませんが、この変が臭そうです。

RyodoTanaka, 2016-02-12:
@mum254 
ありがとうございます。とても勉強になります！

因みに、落ちてる関数は目星をつけて、printfデバッグで見つけましたｗ
目星をつけると言っても、コンストラクタの中で一個づつ見ていっただけなんですけど。。。

いずれにせよ、ありがとうございました。

RyodoTanaka, 2016-02-12:
@mum254 
@Ry0 
`fix-moveit_simple_grasp_server`ブランチですが、 @Ry0 が新たにリポジトリをフォークしてきてくれたので、削除したいと思います。
また、Motomanではまだ実装していませんが、問題は解決したので、このissueを閉じたいと思います。 
ありがとうございました！

closed 2016-02-12 by RyodoTanaka