Nishida-Lab/motoman_project #56: octomapがRviz上でうまく確認できない問題について

RyodoTanaka, 2016-02-12:
# octomapが更新されない問題について
motomanでmoveitを動かす際、octomapが更新されないという問題があります。
これは、Rviz上でoctomapの更新が確認できないというものです。
Rviz上のoctomapを更新するためには、PnanningSceanのRvizプラグインを立ち上げ直す必要がありました。

# 気づいた点・おそらく原因
Rviz上では見えていないのですが、一度GazeboとMoveitを起動させたあと、Gazebo上に新たな障害物を出現させ、プランニングを行ってみると、kinectで見えているところに関してはきちんと避けているようでした。
また、Rviz上ではoctomapに映らないのですが、本来は障害物があるところをゴールにするとプランニングに失敗します。
以上のことから、motomanに関しては、Rvizの表示上で確認できないだけであって、プランナーにはcollision情報がキチンと提供されていると考えることができます。
また、この症状は、motomanだけcurrent姿勢が更新されないことに酷似していると感じます。
故に、原因はmoveitのvisualisationとmotomanに何か齟齬がある為だと考えられます。

### 追加点
モーションプランを実行した際に、kinectで見えていない部分にロボットが衝突し、octomapが変わるようなことがありました。
その際に、以下のようなメッセージが表示され、プランニングの実行が途中で中止されました
```bash
[ INFO] [1455283149.001766923, 159.349000000]: ABORTED: Solution found but the environment changed during execution and the path was aborted
```
このメッセージからもわかるように、Rvizには表示がありませんが、プランナーと実行プログラムには、octomapの情報がリアルタイムで届いていると考えられます。

# 今後について
以上より、おそらくではありますが、原因箇所が少しづつ判明してきました。
今後は、moveitのvisual部分を中心にデバッグしたいと思います。

mum254, 2016-02-14:
すみません、octomapが更新されない件、もう少し教えていただけますか？私が理解できておらず^^;

当方の環境だと、RvizのPointCould2で /kinect2/hd/points を可視化すると、下記の状況でその情報がちゃんと更新されているようなのです。

- ロボットを動かしたとき
 - points がロボットの追従する
- gazeboで新しい物体をkinectの視野内に入れたとき（台の外、台の上どちらでも）
 - 新しい物体が存在するところにpointsが出現する

これ以外の現象を観察して、更新されていないという問題が発生しているのでしょうか？

それとも、そもそもoctomapとは、/kinect2/hd/pointsとは異なるものを指しており、その情報を見て更新されていないという問題が発生するのでしょうか？もしそうである場合、どのような操作をすればRviz上で確認できるか教えていただけますか？

すみませんが、よろしくお願いします。

RyodoTanaka, 2016-02-15:
@mum254 
お返事が遅くなり、申し訳ありません。
現在問題になっているのは、/kinect2/hd/pointsではなく、Moveit!をRviz上で立ち上げた時に出現するボクセル状グリッド（Octomap）が更新されないという点です。
つまり、
 > octomapとは、/kinect2/hd/pointsとは異なるものを指しており、その情報を見て更新されていないという問題が発生する

ということです。
この症状を確認する方法は以下の通りです。
1. sia5 + kinect のGazeboを起動
2. pnanning_executionの立ち上げ

このとき表示されるボクセルグリッドが、/kinect2/hd/pointsと同じように見た目には更新されないということです。

以上、よろしくお願いします。

mum254, 2016-02-15:
@RyodoTanaka 
ありがとうございます。助かります。ちなみに、下記の手順で良いですか？
`roslaunch motoman_gazebo sia5_nishida_lab.launch`
`roslaunch motoman_sia5_moveit_config moveit_planning_execution.launch`

それから、`planning_execution`とは、何を指しますか？ `move_group`ノードを起動した時に有効になる何かと解釈してよいでしょうか？ ※すみません、探してみたのですが、見つけられませんでした。

また、Rvizで表示させるためには、具体的にどの項目にチェックを入れればよいでしょうか？デフォルトの設定だと、`/kinect2/hd/points`しか表示されないようで^^;

[image]

よろしくお願いします。

RyodoTanaka, 2016-02-16:
@mum254 
わかりづらい説明、失礼しました。
起動の手順についてご説明します。
1. `roslaunch motoman_gazebo sia5_nishida_lab.launch`
2. `roslaunch motoman_moveit sia5_nishida_lab.launch`

以上の操作で、以下のような状況をRvizで確認できるはずです。
この時に表示されるボクセルグリッド（octomap）が、`/kinect2/hd/points`と同じように更新されなくて困っている、というのが現在の問題です。よろしくお願いします。
[image]

RyodoTanaka, 2016-02-16:
@mum254 
先ほどの情報に追加ですが、
最新の`master`ブランチにて、
sia5 + kinectv2 のみのモデルも追加しました。
以下の手順でloanchでき、このモデルでも同様の問題が確認できます。
また、こちらのモデルの方が軽いので、もしよければこちらでご確認ください。
以下に手順と動作画面を示します。
1. `roslaunch motoman_gazebo sia5_with_kinectv2.launch`
2. `roslaunch motoman_moveit sia3_with_kinectv2_moveit_planning_execution.launch`
上記コマンドで、以下のような状況が確認できるはずです。
[image]
以上、よろしくお願いします。

Ry0, 2016-02-16:
Learning_ROS_for_Robotics_Programming_2nd_editionで正しくOctomapの更新が行われている動画です．この状況をどうしてもmotomanなど自分たちのところで再現できていません．
https://youtu.be/muTJ-VZzkY8
move_groupノードを立ち上げたあとに物体を追加しても，きちんと更新されています．

RyodoTanaka, 2016-02-16:
@mum254 
@Ry0 
# 動きました！！！！！！！！！！
まずは、動いた様子をアップロードしましたので、こちらからご覧ください。
[image]

### 原因
Rviz上のMotionPlanningプラグインにおいて、`Planning Scene Topic`という項目があるのですが、このトピック名が`/move_group/monitored_planning_scene`になっていないことが原因でした。因みに、うまく動かなかった時のトピック名は`/planning_scene`です。
これにより、
1. Octomapの更新ができるようになった。
2. Currentが更新されるようになった。
上記二つの問題が一度に解決されました！
なので、本issueの最初で述べていた、
> また、この症状は、motomanだけcurrent姿勢が更新されないことに酷似していると感じます。
故に、原因はmoveitのvisualisationとmotomanに何か齟齬がある為だと考えられます。

という推理はあたっていたということです。

#### なぜ、原因究明に手間取ったのか
今回の原因であったトピックですが、`/planning_scene`トピックには、情報が入っていないわけではなく、MotionPlanningプラグインが起動した時のScene情報が入っており、それが原因で発見が遅れてしまいました。

### おわりに
問題が完全に解決しました！これでROSBOOK_ARMと同じ環境を手に入れることができたので、次はMotomanに平行チャックを付けた時と、ジャミンググリッパ（プラグインを書く必要あり）を付けた時で色々と比べてみたいと思います。
ご協力ありがとうございました！

mum254, 2016-02-16:
@RyodoTanaka 
ありがとうございます！こちらでも確認しました！
いやー，私も今日帰ってからずっとこれやっててさっぱりだったんです．すっきりしました！

これは設定のミスなんじゃないかって，launch ファイルやyaml を全部ひっくり返して見てましたが，間違いが見つからないはずです．

ファイルのミスが見当たらないので，見た目側が問題なのかなーとか思いつつ，ちょうど下記ように整理していたとろで，更新がかかりました．せっかくここまで書いたので，情報共有のために投下しておきますね．^^;

- octmap自体は更新されていて，表示だけされてないということは分かった．
- Rviz上の`Motion Planning`の`Status`タブで`rosbook_arm`では`Change goal state`が更新されているのに，`motoman`では更新されていないことがおかしいと思った．
- `rostopic list | grep plan` で出てきたトピックが怪しそうなので，`rosbook_arm`と`motoman`それぞれについて，`rostopic echo` をやって違いを見つけようとした．
- `rostopic echo /move_group/planning_scene_monitor/parameter_updates` をすると，`publish_planning_scene_hz`の値が4.0となっており，`rosbook_arm`では確かにコレくらいの周期で更新されているのに，`motoman`では更新されていない．
 - `planninig_scene_hz`を`publish`している先がどこかにあるはずで，ここの不一致が疑われるが，これはどこなんだろうなぁ？

てな感じで，私は迷子になっていました^^;　お見事です！

この辺り，仕様のドキュメントがないんですかね．google group でも相当悩んでるっぽかったんで，意外とみんな気づいていないかもしれません．
https://groups.google.com/forum/#!topic/moveit-users/MFXZWp7_iSQ

とにかく，とても勉強になりました！

RyodoTanaka, 2016-02-16:
@mum254 
ありがとうございます！
私も、全部ひっくり返して確認していたのですが、何もわからず、途方にくれていました！
いずれにせよ、良かったです。
また、デバッグ手法の共有、ありがとうございます。
勉強になりました。

@mum254 
@Ry0 
ということで、本件は解決したので、閉じたいと思います。
ありがとうございました！

closed 2016-02-16 by RyodoTanaka