Nishida-Lab/motoman_project #68: openraveのコンパイルエラー

Ry0, 2016-05-16:
Mastering ROS for Robotics Programmingの手順にしたがってopenraveのインストールを試みましたが，

Configureの段階でエラーが出てしまいます．
エラーがでた箇所は以下の通りです．

[image]

```bash
CMake Warning at CMakeLists.txt:478 (find_package):
  By not providing "FindASSIMP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ASSIMP", but
  CMake did not find one.

  Could not find a package configuration file provided by "ASSIMP" with any
  of the following names:

    ASSIMPConfig.cmake
    assimp-config.cmake

  Add the installation prefix of "ASSIMP" to CMAKE_PREFIX_PATH or set
  "ASSIMP_DIR" to a directory containing one of the above files.  If "ASSIMP"
  provides a separate development package or SDK, be sure it has been
  installed.
```

```bash
CMake Warning at CMakeLists.txt:709 (find_package):
  By not providing "FindFPARSER.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "FPARSER", but
  CMake did not find one.

  Could not find a package configuration file provided by "FPARSER"
  (requested version 4.4.3) with any of the following names:

    FPARSERConfig.cmake
    fparser-config.cmake

  Add the installation prefix of "FPARSER" to CMAKE_PREFIX_PATH or set
  "FPARSER_DIR" to a directory containing one of the above files.  If
  "FPARSER" provides a separate development package or SDK, be sure it has
  been installed.
```

```baｓｈ
checking for module 'fcl'
  package 'fcl' not found
Could not find FCL. Please install FCL (https://github.com/flexible-collision-library/fcl)
```

こんな状態なので，Generateしたあと，makeしてもコンパイルエラーが起きていまいます．
仮想環境を使ってクリーンな環境でも試しましたが，同じ内容で詰まってしまい困っています．

RyodoTanaka, 2016-05-16:
@Ry0 @MoriKen254 
私も同じようにやってみたところ、`cmake-gui`で`Generate`を押した時点で以下のエラーが出ました。

```bash
Compiling OpenRAVE Version 0.9.0, soversion=0.9
Using cmake version 2.8.12
Using double precision
detected system processor: x86_64
installing to /usr/local, python exe /usr/bin/python, python subdir is lib/python2.7/dist-packages, program suffix is '0.9'
Boost version: 1.54.0
Found the following Boost libraries:
  regex
  filesystem
  system
  python
  thread
  iostreams
  date_time
found boost version: 105400
Targetting Unix Makefiles
Using /usr/bin/c++ compiler.
Could NOT find GMP
Could NOT find MPFR
MPFI needs GMP and MPFR
Using local crlibm
compiling with double extended (long double) floating point
CMake Warning at CMakeLists.txt:478 (find_package):
  By not providing "FindASSIMP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ASSIMP", but
  CMake did not find one.

  Could not find a package configuration file provided by "ASSIMP" with any
  of the following names:

    ASSIMPConfig.cmake
    assimp-config.cmake

  Add the installation prefix of "ASSIMP" to CMAKE_PREFIX_PATH or set
  "ASSIMP_DIR" to a directory containing one of the above files.  If "ASSIMP"
  provides a separate development package or SDK, be sure it has been
  installed.

checking for module 'assimp'
  found assimp, version 3.0.1264
assimp version 3.0.1264, , /usr/include/assimp
WARNING: you are using the obsolete 'PKGCONFIG' macro, use FindPkgConfig
libxml2 found
checking for module 'minizip'
  package 'minizip' not found
compiling local ann library
CMake Warning at CMakeLists.txt:709 (find_package):
  By not providing "FindFPARSER.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "FPARSER", but
  CMake did not find one.

  Could not find a package configuration file provided by "FPARSER"
  (requested version 4.4.3) with any of the following names:

    FPARSERConfig.cmake
    fparser-config.cmake

  Add the installation prefix of "FPARSER" to CMAKE_PREFIX_PATH or set
  "FPARSER_DIR" to a directory containing one of the above files.  If
  "FPARSER" provides a separate development package or SDK, be sure it has
  been installed.

checking for one of the modules 'fparser'
compiling local fparser library
compiling local convexdecomposition library
Geometry Parsing: Collada DOM 2.4.4
Geometry Parsing: Using Open Asset Import Library 3.0.1264
Geometry Parsing: Using ivcon
extracting models to /home/ryodo/Documents/OpenRAVE/openrave/src
Octave installation not found, is 'mkoctfile' in the system path?
Found sympy version '0.7.4.1'
Assuming sympy version 0.7.x or greater
python and boost-python found
Configuring done
Generating done
```
しかし、 @Ry0 のようにそのまま`make`しても特にコンパイルエラーは出ませんでした。
その代わり、
```bash
$ openrave
```
で起動しようとすると、
```bash
[plugindatabase.h:395 Init] /usr/local/lib/openrave0.9-plugins doesn't exist
[plugindatabase.h:577 Create] Failed to create name fcl_, interface collisionchecker
[plugindatabase.h:577 Create] Failed to create name ode, interface collisionchecker
[plugindatabase.h:577 Create] Failed to create name bullet, interface collisionchecker
[plugindatabase.h:577 Create] Failed to create name pqp, interface collisionchecker
```
というエラーが出て、OpenRAVEのGUIが立ち上がりません(エラー内容的に当たり前なのですが...)。
現状ではこんな感じです。よろしくお願いします。

### 動作環境
| | |
| :---: | :---: |
| **PC** | Desktop |
| **Prosessor** |  Intel® Core™ i5-4460 CPU @ 3.20GHz × 4  |
| **RAM** | DDR3 (24GB) |
| **OS** | Ubuntu 14.04 LTS 64bit |
| **Kernel** | 3.13.0-83-generic |

MoriKen254, 2016-05-18:
何かクリティカルな要素が足りていないように見えますが，原因が見えませんね．

取り急ぎ，こちらの環境のメッセージをベタ貼りします．こちらのメッセージとどこが違うのかを，確認してみて下さい．相違点があれば，そこがネックと思われます．

## cmake-gui
[image]

[image]

## Generate

```bash
Compiling OpenRAVE Version 0.9.0, soversion=0.9
Using cmake version 2.8.12
Using double precision
detected system processor: x86_64
installing to /usr/local, python exe /usr/bin/python, python subdir is lib/python2.7/dist-packages, program suffix is '0.9'
Boost version: 1.54.0
Found the following Boost libraries:
  regex
  filesystem
  system
  python
  thread
  iostreams
  date_time
found boost version: 105400
Targetting Unix Makefiles
Using /usr/bin/c++ compiler.
Could NOT find MPFI
Using local crlibm
compiling with double extended (long double) floating point
CMake Warning at CMakeLists.txt:471 (find_package):
  By not providing "FindASSIMP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ASSIMP", but
  CMake did not find one.

  Could not find a package configuration file provided by "ASSIMP" with any
  of the following names:

    ASSIMPConfig.cmake
    assimp-config.cmake

  Add the installation prefix of "ASSIMP" to CMAKE_PREFIX_PATH or set
  "ASSIMP_DIR" to a directory containing one of the above files.  If "ASSIMP"
  provides a separate development package or SDK, be sure it has been
  installed.

checking for module 'assimp'
  found assimp, version 3.0.1264
assimp version 3.0.1264, , /usr/include/assimp
WARNING: you are using the obsolete 'PKGCONFIG' macro, use FindPkgConfig
libxml2 found
checking for module 'minizip'
  package 'minizip' not found
compiling local ann library
CMake Warning at CMakeLists.txt:702 (find_package):
  By not providing "FindFPARSER.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "FPARSER", but
  CMake did not find one.

  Could not find a package configuration file provided by "FPARSER"
  (requested version 4.4.3) with any of the following names:

    FPARSERConfig.cmake
    fparser-config.cmake

  Add the installation prefix of "FPARSER" to CMAKE_PREFIX_PATH or set
  "FPARSER_DIR" to a directory containing one of the above files.  If
  "FPARSER" provides a separate development package or SDK, be sure it has
[... 463 lines cut ...]
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/libconfigurationcache.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/openravepy_configurationcache.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/libdualmanipulation.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/libgrasper.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/libikfastsolvers.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/liblogging.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/liboderave.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/libpqprave.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/libqtcoinrave.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/librmanipulation.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/librplanners.so
-- Up-to-date: /usr/local/share/openrave-0.9/plugins/libtextserver.so
```

## openrave
下記メッセージ後，起動します．

```bash
[plugindatabase.h:929 _SysLoadLibrary] /usr/local/share/openrave-0.9/plugins/openravepy_configurationcache.so: openravepy_int.so: cannot open shared object file: No such file or directory
[plugindatabase.h:857 _LoadPlugin] failed to load: /usr/local/share/openrave-0.9/plugins/openravepy_configurationcache.so
[textserver.h:403 main] text server listening on port 4765
```

## `/usr/local/lib/`の構成
[image]

RyodoTanaka, 2016-08-01:
@MoriKen254 
@Ry0 
以下のURLに載っている方法でインストールできました。
上記エラーの原因がよくわからなかったのが腑に落ちませんが、とりあえず解決策を見つけたので載せておきます。
https://scaron.info/teaching/installing-openrave-on-ubuntu-14.04.html

MoriKen254, 2016-08-03:
@RyodoTanaka 
ありがとうございます^^
インストール後，このリポジトリで動きましたか？

Ry0, 2016-11-28:
久しぶりにコンパイルに試みたところ成功したので報告しておきます．

```
git clone --branch latest_stable https://github.com/rdiankov/openrave.git
```

* ブランチ名： latest_stable
* バージョン確認： [9c79ea2](https://github.com/rdiankov/openrave/commit/9c79ea260e1c009b0a6f7c03ec34f59629ccbe2c)

相変わらず，`cmake ..`の段階で以下のような警告？エラー？がでますが，コンパイルは成功します． 

```bash
  Could not find a package configuration file provided by "ASSIMP" with any
  of the following names:

    ASSIMPConfig.cmake
    assimp-config.cmake
```

```bash
  Could not find a package configuration file provided by "FPARSER"
  (requested version 4.4.3) with any of the following names:

    FPARSERConfig.cmake
    fparser-config.cmake
```

手順は，
https://scaron.info/teaching/installing-openrave-on-ubuntu-14.04.html
にしたがってインストールしました．

motoman_project [Wikiページ](https://github.com/Nishida-Lab/motoman_project/wiki/OpenRAVE%E3%81%AEIKFast%E3%81%A7Motoman%E7%94%A8%E3%81%AEMoveIt!%E3%83%97%E3%83%A9%E3%82%B0%E3%82%A4%E3%83%B3%E3%82%92%E4%BD%9C%E6%88%90%E3%81%99%E3%82%8B)の通りに動作することを今更ながら確認しました．

closed 2016-11-28 by Ry0