Nishida-Lab/motoman_project #87: 一番はじめの catkin_make について (半解決済み)

TakakiNishio, 2017-04-20:
本 issue は私個人の PC にインストールしている環境との干渉が原因の可能性があります．他の方が成功していたらすみません．

## 問題点
wiki ページを参考にして自分の PC に motoman のシミュレータ環境を構築しようとしたところ，iai_kinect 関連のエラーが発生し，catkin_make が通りませんでした．該当箇所のディレクトリを削除することで catkin_make は通るようになりました．パソコン内でのシミュレーションには必要ない箇所と判断した場合の解決策です．

## 環境
* Ubuntu 14.04
* ROS Indigo

## 状況と解決策
catkin_make を実行したところ，下に示すエラーが発生しました (エラーメッセージは発生箇所のみ抜粋しています)．解決策としては，motoman_project 内の /motoman_tools 内のディレクトリ "iai_kinect2" をまるごと削除すれば catkin_make は通るようになります．
```
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function 'bool Kinect2Bridge::initPipeline(const string&, int32_t)':
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
                            ^
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected ';'
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected type-specifier
       packetPipeline = new libfreenect2::CudaKdePacketPipeline(device);
                            ^
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected ';'
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function 'bool Kinect2Bridge::initPipeline(const string&, int32_t)':
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
                            ^
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected ';'
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected type-specifier
       packetPipeline = new libfreenect2::CudaKdePacketPipeline(device);
                            ^
/home/nishio/Documents/motoman_ws_origin/src/motoman_tools/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected ';'
make[2]: *** [motoman_tools/iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] Error 1
make[1]: *** [motoman_tools/iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [motoman_tools/iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] Error 1
make[1]: *** [motoman_tools/iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

RyodoTanaka, 2017-04-20:
@TakakiNishio 
原因はlibfreenect2関係にありそうですが，Travisに登録しているlibfreenect2のやり方だと，

```bash
- yes | sudo apt-add-repository ppa:floe/libusb && sudo apt-get update -qq && sudo apt-get install -y libusb-1.0-0-dev
  - sudo apt-get install -y libturbojpeg libjpeg-turbo8-dev
  - sudo dpkg -i debs/libglfw3*deb; sudo apt-get install -f; sudo apt-get install -y libgl1-mesa-dri-lts-vivid
  - yes | sudo apt-add-repository ppa:pmjdebruijn/beignet-testing; sudo apt-get update -qq; sudo apt-get install -y beignet-dev;
  - git clone https://github.com/OpenKinect/libfreenect2.git
  - cd libfreenect2
  - mkdir build && cd build
  - cmake ..
  - make
  - sudo make install
  - sudo ldconfig
  - cd ~
```

です．
以上の方法でlibfreenect2をインストールしてもだめな場合は，もう少し原因を探る必要があります．

TakakiNishio, 2017-04-20:
回答ありがとうございます．Travis に登録されている libfreenect2 のインストール方法を試しましたが，同じエラーになりました．当面はこのまま進めていきます．私個人の環境が原因かもしれないので，自分でも探ってみます．

RyodoTanaka, 2017-04-20:
@TakakiNishio 
了解です．
他の方もエラー起きたら報告をおねがいします．

RyodoTanaka, 2017-07-08:
@TakakiNishio 
これ，解決したかも知れません．
同じようなエラーを起こしている人を見つけました．
https://github.com/code-iai/iai_kinect2/issues/369
で，解決方法ですが，
1. libfreenectを最新の`master`ブランチでpullし直す
2. libfreenectの`make`と`make install`を行う
3. もう一回`catkin_make` or `catkin build`

私の家の環境では，一度目に同じエラーが出て，上記の修正後はうまくコンパイルが通りました．(`catkin build`で)
もし，必要になったら or 暇だったら検証してみてください．

TakakiNishio, 2017-07-08:
@RyodoTanaka 
回答ありがとうございます．
提示していただいた解決方法で無事 `catkin build` が通りました．
ご協力ありがとうございました！ m(_ _)m

closed 2017-07-08 by TakakiNishio