Nishida-Lab/motoman_project #88: catkin_make時のerror

shimoe, 2017-04-20:
# 問題点
wikiのインストール手順に従ってインストールしていた所、以下のようなエラーに出会いました。
pcl関連のエラーのようですが解決策がわかりません。

## 環境
Ubuntu14.04
ROS indigo
Kernel 3.13.0-116-generic

# エラーコード
[ 26%] Building CXX object motoman_project/motoman_recognition/motoman_euclidean_cluster/CMakeFiles/euclidean_cluster_node.dir/src/euclidean_cluster_node.cpp.o
In file included from /home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/euclidean_cluster.cpp:2:0:
/home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/../include/euclidean_cluster.hpp:19:55: fatal error: pcl/features/moment_of_inertia_estimation.h: そのようなファイルやディレクトリはありません
 #include <pcl/features/moment_of_inertia_estimation.h>
                                                       ^
compilation terminated.
In file included from /home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/euclidean_cluster_node.cpp:1:0:
/home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/include/euclidean_cluster.hpp:19:55: fatal error: pcl/features/moment_of_inertia_estimation.h: そのようなファイルやディレクトリはありません
 #include <pcl/features/moment_of_inertia_estimation.h>
                                                       ^
compilation terminated.

RyodoTanaka, 2017-04-20:
@shimoe 
PCLのインストールは行いましたか？

shimoe, 2017-04-20:
@RyodoTanaka 
していなかったようなので
[installer.bash](https://github.com/RyodoTanaka/my_ubuntu_setup/blob/master/PCL/installer.bash)
を用いてインストールしましたが、再び以下のようなエラーコードが出ました。

[ 68%] Building CXX object motoman_tools/iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
In file included from /home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/euclidean_cluster.cpp:2:0:
/home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/../include/euclidean_cluster.hpp:19:55: fatal error: pcl/features/moment_of_inertia_estimation.h: そのようなファイルやディレクトリはありません
 #include <pcl/features/moment_of_inertia_estimation.h>
                                                       ^
compilation terminated.
In file included from /home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/euclidean_cluster_node.cpp:1:0:
/home/shimoe/motoman_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/include/euclidean_cluster.hpp:19:55: fatal error: pcl/features/moment_of_inertia_estimation.h: そのようなファイルやディレクトリはありません
 #include <pcl/features/moment_of_inertia_estimation.h>
                                                       ^
compilation terminated.

RyodoTanaka, 2017-04-20:
@shimoe 
PCLをインストール後、

- `<catkin_ws>/build`と`<catkin_ws>/devel`を削除

した後，コンパイルをしても同様ですか？

shimoe, 2017-04-20:
@RyodoTanaka 
はい、同様のエラーコードが出ました。

RyodoTanaka, 2017-04-20:
@shimoe 
oh...
ちょっと時間をくださいm(_ _)m

RyodoTanaka, 2017-04-20:
@shimoe 
こんなissueを発見しました．
http://www.pcl-users.org/Moment-of-Inertia-include-error-td4041827.html
上記issueでは，PCL-ROS依存のヘッダーであるかもとの結論に至っています．

そこで，

```bash
$ sudo apt-get install ros-indigo-pcl-ros
```

を実行後も同様のエラーが出るかどうか調べてみてくださいm(_ _)m

あと確認ですが，インストール時に

1. `catkin_ws`に`motoman_project`をclone
2. wstool で必要パッケージを追加
3. `rosdep install -i -y -r --from-paths src --ignore-src`を`<catkin_ws>`で実行

という手順は踏んだ上でエラーが起きているということでOKですか？

shimoe, 2017-04-20:
@RyodoTanaka 
はい
その手順を踏んだ上で上記のエラーが起きてます

RyodoTanaka, 2017-04-20:
@shimoe 
私の環境では，
`/usr/include/pcl-1.7/pcl/features/moment_of_inertia_estimation.h`
にヘッダーがあります．

`/user/include/pcl-***/`
の***の部分はどのようになっていますか?

shimoe, 2017-04-20:
@RyodoTanaka 
pcl-1.7ですね・・・

RyodoTanaka, 2017-04-20:
@shimoe 
Nooooooooooooooooooooooooooooooh!
もう少し時間をくださいm(_ _)m

RyodoTanaka, 2017-04-20:
@shimoe 
`pcl-1.7.2rc1`以上でないと、問題になっているヘッダーが存在しないというところまで確認しました．
紹介したinstaller.basｈでどのバージョンが引っ張ってこられるのかを確認してみます．

shimoe, 2017-04-20:
@RyodoTanaka 
`/usr/include/pcl-1.7/pcl/features/`
の中を見たところ
`moment_of_inertia_estimation.h`
が存在しませんでした。

似たようなヘッダファイルとして
`moment_invariants.h `
があります。

RyodoTanaka, 2017-04-20:
@shimoe 
調べたところ，引っ張って来られるpclのバージョンは1.7.2のようでした
https://launchpad.net/%7Ev-launchpad-jochen-sprickerhof-de/+archive/ubuntu/pcl/+index?field.series_filter=trusty

```bash
$ sudo apt-get install libpcl-1.7-all 
$ cd <catkin_ws>
$ rm -rf <catkin_ws>/build
$ rm -rf <catkin_ws>/devel
$ catkin_make
```

を行っても同様ですか？

shimoe, 2017-04-20:
@RyodoTanaka 
ダメみたいですね・・・
一応`pcl-1.7`の中も確認しましたが存在しませんでした。

RyodoTanaka, 2017-04-20:
@shimoe 
うーん
一番手っ取り早い解決策は、ソースコードを直接cloneしてきてインストールする方法です．
ただ、Travisでエラーが起きていない以上，なぜこのような現象が起きるのか調べる必要があります．

shimoe, 2017-04-22:
@RyodoTanaka 
ソースコードを直接インストールして来てから
apt-get update及びapt-get uograde を行った後、ビルドしたら通りました。

回答有り難うございました

closed 2017-04-22 by shimoe