Nishida-Lab/motoman_project #89: catkin_makeについて

nakaomasato, 2017-04-21:
最初のインストールで下記のようなエラーが出ました（対象箇所抜粋）

WARNING: package "motoman_euclidean_cluster" should not depend on metapackage "jsk_visualization" but on its packages instead
WARNING: package "motoman_cable_removal" should not depend on metapackage "jsk_visualization" but on its packages instead
WARNING: Package 'motoman_config' is deprecated (This package will be removed in ROS Indigo.  The configuration data included in this

CMake Error at motoman_tools/iai_kinect2/kinect2_bridge/CMakeLists.txt:22 (find_package):
  Could not find a package configuration file provided by "freenect2" with
  any of the following names:

    freenect2Config.cmake
    freenect2-config.cmake

  Add the installation prefix of "freenect2" to CMAKE_PREFIX_PATH or set
  "freenect2_DIR" to a directory containing one of the above files.  If
  "freenect2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/nakao/motoman_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nakao/motoman_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

RyodoTanaka, 2017-04-21:
@nakaomasato 
これは，freenect2のインストールができていないために起きます.
freenect2のインストールが必要です．

```bash
yes | sudo apt-add-repository ppa:floe/libusb && sudo apt-get update -qq && sudo apt-get install -y libusb-1.0-0-dev
sudo apt-get install -y libturbojpeg libjpeg-turbo8-dev
sudo dpkg -i debs/libglfw3*deb; sudo apt-get install -f; sudo apt-get install -y libgl1-mesa-dri-lts-vivid
yes | sudo apt-add-repository ppa:pmjdebruijn/beignet-testing; sudo apt-get update -qq; sudo apt-get install -y beignet-dev;
git clone https://github.com/OpenKinect/libfreenect2.git
cd libfreenect2
mkdir build && cd build
cmake ..
make
sudo make install
sudo ldconfig
cd ~
```

以上を実行してみてください

nakaomasato, 2017-04-21:
sudo dpkg -i debs/libglfw3*deb;
を入力すると

dpkg: アーカイブ debs/libglfw3*deb の処理中にエラーが発生しました (--install):
 アーカイブにアクセスできません: そのようなファイルやディレクトリはありません
処理中にエラーが発生しました:
 debs/libglfw3*deb

と出ました

shimoe, 2017-04-21:
@nakaomasato 
[libfreenect2](https://github.com/OpenKinect/libfreenect)をクローンしてからビルドしてみて下さい。

以下のコマンドを実行すればできます
`git clone http://github.com/OpenKinect/libfreenect2.git`
`cd libfreenect2/'
`mkdir build`
`cd build/'
`cmake ..'
`make'
`sudo make install`

nakaomasato, 2017-04-21:
田中さんが答えたプログラムの5行目に、クローンのコマンドがあったので特に変わりませんでした。

nakaomasato, 2017-04-26:
cd catkin_ws
mkdir lib && cd lib
git clone https://github.com/PointCloudLibrary/pcl.git
cd pcl
mkdir build && build
cmake ..
make -j2
sudo apt-get update
sudo apt-get upgrade
make
cd catkin_ws
rm -rf motoman_ws/build
rm -rf motoman_ws/devel
catkin_make
をするとcatkin makeが成功しました。

closed 2017-04-26 by nakaomasato