Nishida-Lab/motoman_project #91: motomanのcalibrationに関して

Yuji-Ninomiya, 2017-04-25:
motoman_project のコマンドチートを参照しながらcalibrationを行おうとしたのですが、gazeboにロボットモデルを登場させる以下のコマンド
```
roslaunch motoman_gazebo sia5_with_dhand_and_multikinect.launch
```
を実行するところで下記のエラーが生じました（motomanおよびkinectのは接続済）。
```
[ERROR] [1493107510.710925494, 2.346000000]: Lookup would require extrapolation into the past.  
Requested time 0.517000000 but the earliest data is at time 1.005000000, when looking up transform from 
frame [kinect_first_rgb_optical_frame] to frame [kinect_second_rgb_optical_frame]
```
gazebo上ではロボットが表示されましたが、RVizではロボットモデルがワールド座標原点付近に固まったように表示されています。
TFの箇所に ```Status: Warn```が出ているのでそこがうまくいってないように思うのですが、アドバイスをいただけると幸いです。
よろしくお願いします。

## 現状
Fixed Frame:: world
Robot Model:: Status: Error
TF:: Status: Warn

[image]

RyodoTanaka, 2017-04-26:
@Yuji-Ninomiya 
返事が遅くなってしまい，申し訳ありません．
まず，
> motoman_project のコマンドチートを参照しながらcalibrationを行おうとしたのですが、gazeboにロボットモデルを登場させる以下のコマンド
> ```
> roslaunch motoman_gazebo sia5_with_dhand_and_multikinect.launch
> ``` [...]

この部分に関しては，問題ありません．私の環境でも起動時に表示されるエラーですが，動作に影響のあるものではありません．

続いて，ロボットモデルがうまく表示されない問題ですが，これは`robot_state_publisher`もしくは，`joint_state_publisher`が動いていないために起きる現象です．　
Gazeboの起動時にこれが起きるのは，うまくコントローラが読み込まれていないことが原因であることが多いのですが，Gazeboの起動画面コントローラに関するエラーメッセージは出ていませんか？

ただそもそものお話になってしまうのですが，コマンドチートシートに載っているcalibrationは，Kinectの外部パラメータ（位置・姿勢）が未知の状態であり，Kinectに関する情報の載っていないURDFを読み込んでいる状態で行うものです．つまり，実機のみでしか使用できないものです．なので，投稿されているような，Gazebo上でこれを行うことはできないのですが，大丈夫でしょうか？

Yuji-Ninomiya, 2017-04-28:
回答ありがとうございます。お返事が遅れて申し訳ありません。

ロボットモデル表示の問題についてですが、次のような警告文が表示されています。
```
WARNING: disk usage in log directory [/home/motoman/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
....
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/pose/local/info, 
deleting message. This warning is printed only once.
```
コントローラに関するエラーではないような気がしますが、一つ目のエラーは私の環境でgazeboを起動する際には見られませんでした。
```rosclean```について、ROSのログを消去するという説明とともに「ファイルをディスクから消去するものであるため注意して実行せよ」との記述がありましたが、これは実行しても大丈夫なのでしょうか？

今回私の実行しようとしたcalibrationの説明が、実機によるものとGazeboを用いてのシミュレーションとを混同してしまっていたため投稿内容があやふやになってしまっていました。申し訳ありません。
コマンドチートは参考にしているだけなのでその点は大丈夫です。

RyodoTanaka, 2017-04-28:
@Yuji-Ninomiya @Nishida-Lab/all 

> ロボットモデル表示の問題についてですが、次のような警告文が表示されています。
> ```
> WARNING: disk usage in log directory [/home/motoman/.ros/log] is over 1GB.
> It's recommended that you use the 'rosclean' command.
> ....
> Warning [...]

このエラーについては，書いてある通りにファイルを削除してください．また，このエラーが出ていても，パッケージが動かなくなったりすることはありませんし，ROSをしばらく使っていて，logファイルが溜まってくると必ず起きる現象です．

続いて，

> 今回私の実行しようとしたcalibrationの説明が、実機によるものとGazeboを用いてのシミュレーションとを混同してしまっていたため投稿内容があやふやになってしまっていました。申し訳ありません。
> コマンドチートは参考にしているだけなのでその点は大丈夫です。

混同していないことはわかりましたが，最初に起きたエラーがなぜ起きたのか考察するための情報がもう少し必要です．
モデルがうまく起動されなかった時のPCの環境，実行したコマンド，エラーメッセージ全文（まだ残ってるなら）を記入してください．また，このissueに限ったことではありませんが，問題が起きた時には，

- PCの環境（スペック・kernel version・ROS version等，臨機応変に）
- 実行したコマンド（つまりは，何をやったのかを正確にすべて記入）
- 得られた結果（エラーメッセージ，動くor動かなかった等の情報）

以上の情報を記入していた方が答えにたどり着きやすいです．
面倒だとは思いますが，よろしくお願いしますm(_ _)m

Yuji-Ninomiya, 2017-04-29:
今回実行したコマンド、およびPCの環境は以下のとおりです。
```
rosaddress server
roslaunch motoman_control sia5_with_dhand_and_multi_kinect_streaming.launch
roslaunch motoman_control sia5_real_control.launch 
roslaunch motoman_gazebo sia5_with_dhand_and_multikinect.launch
rviz
```
### PC環境
Linux 64bit
Ubuntu 14.04.5 LTS
kernel: 3.13.0-106-generic
ROS: indigo

```roslaunch motoman_control sia5_with_dhand_and_multi_kinect_streaming.launch```、および```roslaunch motoman_control sia5_real_control.launch```を実行しない状態だとロボットモデルは正しく表示されましたが、上記の流れでコマンドを実行するとRVizではモデルがワールド座標原点に固まって表示されました。エラーメッセージは先にお伝えしたもの以外表示されておりません。また、先のご指摘にもありました```robot_state_publisher```と```joint_state_publisher```に関してですが、後者の```joint_state_publisher```の起動に関する表示がないようです。ここに問題があるということでしょうか。
よろしくお願いします。

Ry0, 2017-04-29:
@Yuji-Ninomiya
```bash
roslaunch motoman_control sia5_real_control.launch 
```
と
```bash
roslaunch motoman_gazebo sia5_with_dhand_and_multikinect.launch
```
は共存できない．
`sia5_with_dhand_and_multikinect.launch`では
`sia5_sim_control.launch`を呼び出すようになっていて
https://github.com/Nishida-Lab/motoman_project/blob/master/motoman_gazebo/launch/sia5/sia5_with_dhand_and_multikinect.launch#L36
コントローラがシミュレーション用と実機用のものが多重起動されています．
実機の場合，joint_stateは実機から送られてきますので，気にしなくてもいい気がします！

このissueのもともとはキャリブレーションだよね
従来の手動の方法であれば，Gazeboは関係ないはず．
↓コマンドチートシートってこのこと？
https://github.com/Nishida-Lab/motoman_project/wiki/%E3%82%B3%E3%83%9E%E3%83%B3%E3%83%89%E3%83%81%E3%83%BC%E3%83%88%E3%82%B7%E3%83%BC%E3%83%88

Yuji-Ninomiya, 2017-04-30:
@RyodoTanaka @Ry0 

回答ありがとうございます．
コントローラの多重起動が原因だったのですね．

前述のコマンドチートというのは株丹さんのおっしゃったもので間違いありません．
それらひとつひとつのつながりを十分に理解できていなかったために今回のような事態に陥ったと考えられます．ご指摘ありがとうございました．

Yuji-Ninomiya, 2017-05-11:
多重起動は修正したのですが、実機のKinect から点群が取れないという問題が発生しました。
RViz内で確認したところ、PointCloud2 の Transform の箇所に「 TF の取得時間とリクエスト時間がおかしいため、表示できません」的なエラーが出ていました。（エラーのコピペ氏忘れました、すみません。）

しかし、以下のコマンドを 実行することによりこの問題は修正できました。ありがとうございました。

### 解決策
chronyを使ってサーバPCとクライアントPCの時間を同期することで修正
```
クライアントPCへSSH接続（ch3とする）
$ cd ~
$ sudo ./chrony_setup <htp_server_address  or  Server PC adddress>
$ ntpdate -q <htp_server_address  or  Server PC adddress>
```

closed 2017-05-11 by Yuji-Ninomiya