Nishida-Lab/motoman_project #109: Travisのビルドが通らない！！

RyodoTanaka, 2018-01-27:
https://travis-ci.org/Nishida-Lab/motoman_project/builds/334039073
indigoのTravisが通りません（泣）
ros-indigo-moveit まわりのパッケージがapt-getでインストールできていないのが原因です．
でもどうやって直したらいいのか，なんでそもそもapt-getできないのかわかりあませぇん（泣）
今はプライオリティが低いのでとりあえず放置しておきます．
なお，作業ブランチは`ryodo/fix-ci`です．

MoriKen254, 2018-01-27:
公式の`industrial_ci`でもアウトですね．意味不明です．

`rosdep_install`が失敗します．パッケージが軒並み見つからないようです．これはひどい…．こんなことってあるんでしょうか？

一時的に不調なだけならいいんですけど…．少し様子を見て，直ることを祈りましょう．

- https://travis-ci.org/Nishida-Lab/motoman_project/builds/334102420

```
dpkg-query: no packages found matching ros-indigo-industrial-utils
dpkg-query: no packages found matching ros-indigo-geometry-msgs
dpkg-query: no packages found matching ros-indigo-genmsg
dpkg-query: no packages found matching ros-indigo-roslint
dpkg-query: no packages found matching ros-indigo-tf-conversions
dpkg-query: no packages found matching ros-indigo-rospy
dpkg-query: no packages found matching ros-indigo-moveit-commander
dpkg-query: no packages found matching ros-indigo-pcl-msgs
dpkg-query: no packages found matching ros-indigo-rosserial-python
dpkg-query: no packages found matching libeigen3-dev
dpkg-query: no packages found matching ros-indigo-industrial-robot-simulator
dpkg-query: no packages found matching ros-indigo-actionlib-msgs
dpkg-query: no packages found matching ros-indigo-moveit-core
dpkg-query: no packages found matching liblapack-dev
dpkg-query: no packages found matching ros-indigo-jsk-topic-tools
dpkg-query: no packages found matching ros-indigo-moveit-ros-visualization
dpkg-query: no packages found matching ros-indigo-control-msgs
dpkg-query: no packages found matching ros-indigo-pcl-ros
dpkg-query: no packages found matching ros-indigo-roscpp
dpkg-query: no packages found matching ros-indigo-std-msgs
dpkg-query: no packages found matching ros-indigo-rosserial-arduino
dpkg-query: no packages found matching ros-indigo-robot-state-publisher
dpkg-query: no packages found matching ros-indigo-message-generation
dpkg-query: no packages found matching ros-indigo-controller-manager
dpkg-query: no packages found matching ros-indigo-moveit-planners-ompl
dpkg-query: no packages found matching ros-indigo-ros-controllers
dpkg-query: no packages found matching ros-indigo-sensor-msgs
dpkg-query: no packages found matching ros-indigo-cmake-modules
dpkg-query: no packages found matching ros-indigo-tf
dpkg-query: no packages found matching ros-indigo-urdf
dpkg-query: no packages found matching libpcl-1.7-all
dpkg-query: no packages found matching ros-indigo-simple-message
dpkg-query: no packages found matching libpcl-1.7-all-dev
dpkg-query: no packages found matching ros-indigo-gazebo-ros
dpkg-query: no packages found matching ros-indigo-moveit-msgs
dpkg-query: no packages found matching ros-indigo-rviz
dpkg-query: no packages found matching ros-indigo-joint-state-publisher
dpkg-query: no packages found matching ros-indigo-gazebo-ros-control
dpkg-query: no packages found matching ros-indigo-xacro
dpkg-query: no packages found matching ros-indigo-industrial-robot-client
dpkg-query: no packages found matching ros-indigo-rosserial
dpkg-query: no packages found matching ros-indigo-jsk-visualization
dpkg-query: no packages found matching ros-indigo-warehouse-ros
dpkg-query: no packages found matching ros-indigo-pluginlib
dpkg-query: no packages found matching ros-indigo-industrial-msgs
dpkg-query: no packages found matching ros-indigo-moveit-ros-planning-interface
dpkg-query: no packages found matching libboost-all-dev
dpkg-query: no packages found matching ros-indigo-moveit-ros-move-group
dpkg-query: no packages found matching ros-indigo-trajectory-msgs
dpkg-query: no packages found matching ros-indigo-ros-control
dpkg-query: no packages found matching ros-indigo-moveit-ros-perception
dpkg-query: no packages found matching ros-indigo-message-runtime
dpkg-query: no packages found matching ros-indigo-moveit-simple-controller-manager
dpkg-query: no packages found matching ros-indigo-actionlib
dpkg-query: no packages found matching ros-indigo-roslaunch
dpkg-query: no packages found matching ros-indigo-eigen-conversions
dpkg-query: no packages found matching ros-indigo-industrial-utils
dpkg-query: no packages found matching ros-indigo-geometry-msgs
dpkg-query: no packages found matching ros-indigo-genmsg
dpkg-query: no packages found matching ros-indigo-roslint
dpkg-query: no packages found matching ros-indigo-tf-conversions
[... 27 lines cut ...]
dpkg-query: no packages found matching ros-indigo-gazebo-ros
dpkg-query: no packages found matching ros-indigo-controller-manager
dpkg-query: no packages found matching ros-indigo-rviz
dpkg-query: no packages found matching ros-indigo-joint-state-publisher
dpkg-query: no packages found matching ros-indigo-gazebo-ros-control
dpkg-query: no packages found matching ros-indigo-xacro
dpkg-query: no packages found matching ros-indigo-industrial-robot-client
dpkg-query: no packages found matching ros-indigo-rosserial
dpkg-query: no packages found matching ros-indigo-jsk-visualization
dpkg-query: no packages found matching ros-indigo-warehouse-ros
dpkg-query: no packages found matching ros-indigo-pluginlib
dpkg-query: no packages found matching ros-indigo-industrial-msgs
dpkg-query: no packages found matching ros-indigo-moveit-ros-planning-interface
dpkg-query: no packages found matching ros-indigo-moveit-msgs
dpkg-query: no packages found matching ros-indigo-moveit-ros-move-group
dpkg-query: no packages found matching ros-indigo-trajectory-msgs
dpkg-query: no packages found matching ros-indigo-eigen-conversions
dpkg-query: no packages found matching ros-indigo-ros-control
dpkg-query: no packages found matching ros-indigo-message-runtime
dpkg-query: no packages found matching ros-indigo-moveit-simple-controller-manager
dpkg-query: no packages found matching ros-indigo-actionlib
dpkg-query: no packages found matching ros-indigo-roslaunch
dpkg-query: no packages found matching ros-indigo-moveit-ros-perception
dpkg-query: no packages found matching libboost-all-dev
```

MoriKen254, 2018-01-28:
`industrial_ci`でとりあえず通しました．ブランチは`mk/fix-ci`．とりあえず参考程度に．

- https://github.com/Nishida-Lab/motoman_project/tree/mk/fix-ci
- https://travis-ci.org/Nishida-Lab/motoman_project/builds/334340328

`industrial_ci`でビルドがうまくいってしまうのは，引っ張ってくるDokcerイメージに予め`ROS`が入っているからでしょう．`rosdep`に失敗しているので，最新版は取得できていないと思われます．

それと，なぜ大本のtravisのコードで`indigo`の`moveit`周りのパッケージのインストールが失敗しているのかは不明です．やっぱり`rosdep`の調子が悪いのでしょうか？

また，以下の制約はあります．

- `launch test`は未サポート
  - `catkin build`経由で`launch test`を行うと，他パッケージの依存関係を解決できません．
  - `catkin_make_isolated`の制約と思われます．
  - とりあえず`travis`を通すことを第一目標として，これらのテストはコメントアウトしています．

- `rosdep`でパッケージが軒並み見つからないのは変わらず．
  - `rosdep`の調子が悪いのは共通事項．

- `PCL`を手動でインストールしている
  - `pcl_ros`を依存関係に加えておけば，本来は`rosdep`が入れてくれるはず．
  - `PCL`はDockerイメージにはデフォルトでは入っていないので，仕方なく`.pre_process.sh`でインストール用のスクリプトを追加．

130s, 2018-01-28:
部外者・industrial_ci メンテナです :)

> industrial_ciでビルドがうまくいってしまうのは，引っ張ってくるDokcerイメージに予めROSが入っているからでしょう．rosdepに失敗しているので，最新版は取得できていないと思われます．

デフォでは (i.e. `DOCKER_IMAGE` 変数でイメージを与えて無ければ)，ROS は catkin しか入りません ([ここらへん](https://github.com/ros-industrial/industrial_ci/blob/2425bf1f12e36e5372a826b333474685c5183413/industrial_ci/src/docker.sh#L214-L240)でデフォの設定をしてます)．色々入れちゃうと CI の目的に支障がでるので…

					       
https://travis-ci.org/Nishida-Lab/motoman_project/builds/334102420#L810 を見ると rosdep install は成功してて，build に失敗してそうですが，パット見よく分かりません．
```
Errors     << motoman_euclidean_cluster:make /root/catkin_ws/logs/motoman_euclidean_cluster/build.make.000.log
In file included from /root/catkin_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/euclidean_cluster.cpp:2:0:
/root/catkin_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/../include/euclidean_cluster.hpp:19:55: fatal error: pcl/features/moment_of_inertia_estimation.h: No such file or directory
 #include <pcl/features/moment_of_inertia_estimation.h>
                                                       ^
```	
industrial_ci で問題がある場合は https://github.com/ros-industrial/industrial_ci/issues で報告という名の苦情を下さると，結構早く誰かが返信します．

RyodoTanaka, 2018-01-29:
@MoriKen254 @130s 
検証ありがとうございます．

@130s さんのおっしゃっているとおり，rosdepではなく，パッケージのbuild時に失敗してそうだと思います．
ただ， @MoriKen254 さんの指摘なさっているrosdep がうまくいかないという部分は
https://travis-ci.org/Nishida-Lab/motoman_project/builds/334102420#L566
からのログにあるパッケージが見つからないという旨のエラーだと思うのですが，
https://travis-ci.org/Nishida-Lab/motoman_project/builds/334102420#L685
からのログで見つからなかったはずのパッケージをインストールできているようです．
この挙動は一体何なのか正直よくわかりません．

RyodoTanaka, 2018-01-29:
@MoriKen254 
すみません．やっと意味がわかってきました．
@130s さんから指摘頂いている buildが通っていない部分
https://travis-ci.org/Nishida-Lab/motoman_project/builds/334102420#L810
は，PCLのインストールができていないから起きる問題であり，これに対して「BEFORE_SCRIPT」変数からインストールスクリプトを指定することでPCLをインストールしている．
一方，DockerイメージにPCLが入っていないのでスクリプトを書いて対処するというやり方は，このままで良いと思います．
というのも， @130s さんのおっしゃっている

> デフォでは (i.e. DOCKER_IMAGE 変数でイメージを与えて無ければ)，ROS は catkin しか入りません (ここらへんでデフォの設定をしてます)．色々入れちゃうと CI の目的に支障がでるので…

という方針に沿うやり方になると考えるからです．

というわけで未解決な課題は
- launch test ができない
- rosdepの挙動がよくわからない

ということだと考えています．

RyodoTanaka, 2018-01-29:
@MoriKen254 
launch test ができない問題についてもう少し検討した後，マージしたいと考えていますのでよろしくお願いします．

RyodoTanaka, 2018-01-29:
launch test できない問題についてですが，
d9536b34f8141ebd0aac0c53a80a4e6ce4628170 （launch test消す前）にチェックアウトし，手動で
1. catkin build
2. catkin_run_tests

とやってもエラーは出ません...
もちろんすべてゼロからビルドしている状態です．

RyodoTanaka, 2018-01-29:
なんか関係してそうだったので，備忘録のために貼っておきます．
https://github.com/catkin/catkin_tools/issues/139

MoriKen254, 2018-01-29:
@130s 
ご指摘ありがとうございます．メンテナのご登場，大変ありがたいです :)
おっしゃるとおりですね．小生の認識不足で，失礼致しました ^^;

@RyodoTanaka 
諸々了解です．

## launch test 結果について

> 1. catkin build
> 2. catkin run_tests
> とやってもエラーは出ません...

その後，`catkin_test_results`でテスト結果を確認してみましたか？小生の環境で下記の手順を実行したら，失敗して後述のエラーが出ました．

### 手順
1. catkin build
1. catkin run_tests
1. catkin_test_results

### 結果出力
```
build/motoman_control/test_results/motoman_control/roslaunch-check_test_sia5_with_dhand_and_multi_kinect_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_control/test_results/motoman_control/roslaunch-check_test_sia5_with_dhand_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_description/test_results/motoman_description/roslaunch-check_test_sia5_sia5_with_dhand_and_multi_kinect_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_description/test_results/motoman_description/roslaunch-check_test_sia5_sia5_with_dhand_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_gazebo/test_results/motoman_gazebo/roslaunch-check_test_sia5_with_dhand_and_multi_kinect_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_gazebo/test_results/motoman_gazebo/roslaunch-check_test_sia5_with_dhand_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_mh5_support/test_results/motoman_mh5_support/roslaunch-check_test_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_gazebo_nishida_lab_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_with_dhand_and_multi_kinect_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_with_dhand_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_with_jamming_and_kinectv2_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_with_jamming_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_with_kinectv2_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_moveit/test_results/motoman_moveit/roslaunch-check_test_sia5_with_multi_kinectv2_moveit_planning_execution_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_sda10f_support/test_results/motoman_sda10f_support/roslaunch-check_test_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_sia10d_support/test_results/motoman_sia10d_support/roslaunch-check_test_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_sia10f_support/test_results/motoman_sia10f_support/roslaunch-check_test_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_sia20d_support/test_results/motoman_sia20d_support/roslaunch-check_test_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/motoman_sia5d_support/test_results/motoman_sia5d_support/roslaunch-check_test_launch_test.xml.xml: 1 tests, 0 errors, 1 failures, 0 skipped
Summary: 36 tests, 0 errors, 20 failures, 0 skipped
```
## ではどうすればよいのか？
すみません，結論，私には分かりませんでした笑．

1年ほど前に，`CIR-KIT-Unit03`のパッケージを作りましたね．

- http://wiki.ros.org/Robots/CIR-KIT-Unit03

あの時に`kinetic`対応をしたくて，`industrial_ci`で`launch test`を通そうと色々奮闘した時期があったのですが，結果良くわからず，結局`launch test`はコメントアウトしました．

当時，やはり`industrial_ci`のビルドツールを`catkin_make`に切り替えたりできないかなーと思っていたら，次のissueを見かけました．`launch test`に対して同様の問題提起をしています．

- https://github.com/ros-industrial/industrial_ci/issues/3

最終的には`catkin build`で一本化したほうがいいよね？的な流れになったのでしょうか．`launch test`を救う手段が見当たらず．．．

MoriKen254, 2018-01-29:
## 以下参考：`industrial_ci`上での実際のコマンド

さて，`industrial_ci`上で実際には，後述するコマンドが実行されているようです．

- 参考：https://github.com/ros-industrial/industrial_ci/blob/03709b614c85658eac8d0b5f1f00b9eca18f2352/industrial_ci/src/tests/source_tests.sh#L175

ローカルで試す場合は`/root/catkin_ws`を実際のワークスペースにして下さい．

小生のローカルで試したら，こちらでもちゃんと（？）失敗できました．単一のパッケージ内でテスト設計する，という思想なのかなぁと，また（メンテナの前で）勝手に解釈しています．

- 参考：http://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html#building-single-packages

### ビルドコマンド
```bash
catkin build --verbose --catkin-make-args run_tests -- -w /root/catkin_ws --no-deps --no-status motoman_cable_removal motoman_control motoman_demo_msgs motoman_demo motoman_description motoman_gazebo motoman_gripper motoman_moveit motoman_point_cloud motoman_sia5_moveit_plugins motoman_viz_msgs motoman_euclidean_cluster motoman_rviz_plugins motoman_viz -p1 --make-args -j1 --
```

### テストコマンド
``` bash
catkin run_tests -w /root/catkin_ws --no-deps --no-status motoman_cable_removal motoman_control motoman_demo_msgs motoman_demo motoman_description motoman_gazebo motoman_gripper motoman_moveit motoman_point_cloud motoman_sia5_moveit_plugins motoman_viz_msgs motoman_euclidean_cluster motoman_rviz_plugins motoman_viz -p1 --make-args -j1 --
```

### 結果確認

```bash
catkin_test_results --all
```

失敗

```bash
Summary: 30 tests, 0 errors, 19 failures, 0 skipped
```

RyodoTanaka, 2018-01-29:
@MoriKen254 
私の勉強不足でお手数をおかけします．
ありがとうございます．
ビルド，テストコマンド，結果確認
それぞれのコマンドを教えていただいた通りに行ったところ，全く同じ結果が出ました．

industrial_ciの方で行われているビルドではないのであまり意味は無いのですが，
- ビルド
```bash
$ catkin build
```
- テストコマンド
```bash
$ catkin run_tests
```

- 結果確認
```bash
$ catkin_test_results --all
```

- 出力（失敗）
```
Summary: 36 tests, 0 errors, 1 failures, 0 skipped
```
以上のようになりました．
`--no-deps`を初めとするオプションを何もつけていないので当たり前ですが，勉強のために記録を残しておきます．

RyodoTanaka, 2018-01-29:
@MoriKen254 
すでにご存知かもしれませんが，解決のヒントになるかもしれないissueを見つけました．
https://github.com/ros-industrial/industrial_ci/pull/104
上記issue内でさらに，`rosdep`のオプションに関するものも発見しました．
https://github.com/ros-industrial/industrial_ci/pull/180
まだ見つけただけでどうというわけではないのですが，とりあえず記しておきます．

RyodoTanaka, 2018-01-29:
途中報告です．
### 1. .travis.yamlでsub folder指定してない問題
wstoolでいろいろパッケージを引っ張ってきますが，Dockerで使用される`PKGS_DOWNSTREAM`変数にそれらのパッケージ名が追加されていないのがテスト失敗の１原因になっているようです．
指定方法をまだ検証していないのですが（別問題でローカルのテストがうまくいっていないため）下記PRでsub folderが`.travis.yaml`から指定できるようになっているようです．
https://github.com/ros-industrial/industrial_ci/pull/67

### 2. package.xmlのrun_dependに追記し忘れ問題
まだリモートにPushしていないのですが，launchファイルから`motoman_description`からモデルを読み込む必要があるのに，`run_depend`に記入漏れのあるパッケージがいくつか存在しました．
修正後はそれが原因でテスト失敗ということはなくなりました．

### 3. cloneしてる`motoman`パッケージでもテストがコケる問題
現在は，本家の[`motoman`リポジトリ](https://github.com/ros-industrial/motoman/tree/indigo-devel)からForkした[オリジナルの`motoman`パッケージ](https://github.com/Nishida-Lab/motoman)を使用していますが，本家，オリジナル両方共テストでコケてしまいます．まだ原因を解明していない点と，オリジナルのパッケージではTravisも通っているのでもう少し調べてみます．

RyodoTanaka, 2018-01-29:
# とりあえずローカルでコケないようになりました！
sub folderが指定できている前提でビルドを行いました．
前述の通り，wstoolで引っ張ってきているパッケージへのパスは，`.travis.yaml`で指定する必要がありますが，とりあえずはそれに成功しているとして(`PKGS_DOWNSTREAM`変数に格納できていると仮定して)下記の手順でビルドを行いました．

### 1. 一時的な環境変数の設定(`PKGS_DOWNSTREAM`変数の設定)
```bash
export PKGS_DOWNSTREAM="motoman_mh5_support motoman_sda10f_support motoman_sia20d_moveit_config motoman motoman_msgs motoman_sia10d_support motoman_sia20d_support motoman_driver motoman_sda10f_moveit_config motoman_sia10f_support motoman_sia5d_support dhand_control dhand_description dhand_driver dhand_gazebo dhand_msgs dhand_ros_pkg motoman_demo_msgs motoman_moveit motoman_sia5_moveit_plugins motoman_description motoman_cable_removal  motoman_euclidean_cluster  motoman_point_cloud motoman_viz motoman_control motoman_gazebo motoman_rviz_plugins motoman_viz_msgs motoman_demo motoman_gripper motoman_sia5_moveit_config"
```
### 2. ビルド
```bash
catkin build --verbose --catkin-make-args -- -w ./ --no-deps --no-status $PKGS_DOWNSTREAM -p1 --make-args -j`nproc` --
```
@MoriKen254 に紹介頂いたビルド方法だと，`catkin build run_tests`になっており，コンパイルすべき実行ファイルが作成されませんでした．それが原因で，launchテスト結果に
```log
cannot find node [robot_state] in package [motoman_driver]
```
のようなエラーが発生していました．
また，教えていただいた`industrial_ci`でのビルドでは`catkin build run_tests`という記述は確認できなかった([参照](https://github.com/ros-industrial/industrial_ci/blob/03709b614c85658eac8d0b5f1f00b9eca18f2352/industrial_ci/src/tests/source_tests.sh#L171-L186))ので，上記のようなビルド方法にしました．
間違ってたらごめんなさい．

### 2. テストコマンド
```bash
catkin run_tests -w ./ --no-deps --no-status $PKGS_DOWNSTREAM -p1 --make-args -j`nproc` --
```
これは特に変更ありません．

### 3. 結果確認とと出力
```bash
catkin_test_results --all
```
```log
Skipping "src/motoman/motoman_sia10d_support/package.xml": 'tests'
Skipping "src/motoman/motoman_sia10d_support/test/launch_test.xml": 'tests'
Skipping "src/motoman/motoman_sia5d_support/package.xml": 'tests'
Skipping "src/motoman/motoman_sia5d_support/test/launch_test.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_support/package.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_support/test/launch_test.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_support/package.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_support/test/launch_test.xml": 'tests'
Skipping "src/motoman/motoman_msgs/package.xml": 'tests'
Skipping "src/motoman/motoman/package.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/package.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/ompl_planning_pipeline.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/trajectory_execution.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/motoman_sia20d_moveit_controller_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/sensor_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/planning_pipeline.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/fake_moveit_controller_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/motoman_sia20d_moveit_sensor_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia20d_moveit_config/launch/warehouse_settings.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/package.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/ompl_planning_pipeline.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/trajectory_execution.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/sensor_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/planning_pipeline.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/fake_moveit_controller_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/motoman_sda10f_moveit_controller_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/motoman_sda10f_moveit_sensor_manager.launch.xml": 'tests'
Skipping "src/motoman/motoman_sda10f_moveit_config/launch/warehouse_settings.launch.xml": 'tests'
Skipping "src/motoman/motoman_sia10f_support/package.xml": 'tests'
Skipping "src/motoman/motoman_sia10f_support/test/launch_test.xml": 'tests'
Skipping "src/motoman/motoman_mh5_support/package.xml": 'tests'
Skipping "src/motoman/motoman_mh5_support/test/launch_test.xml": 'tests'
Skipping "src/motoman/motoman_driver/package.xml": 'tests'
Skipping "src/motoman_project/motoman_viz_msgs/package.xml": 'tests'
Skipping "src/motoman_project/motoman_recognition/motoman_cable_removal/package.xml": 'tests'
Skipping "src/motoman_project/motoman_recognition/motoman_euclidean_cluster/package.xml": 'tests'
Skipping "src/motoman_project/motoman_recognition/motoman_point_cloud/package.xml": 'tests'
Skipping "src/motoman_project/motoman_demo_msgs/package.xml": 'tests'
Skipping "src/motoman_project/motoman_rviz_plugins/package.xml": 'tests'
[... 77 lines cut ...]
build/motoman_gazebo/test_results/motoman_gazebo/roslaunch-check_test_sia5_with_kinectv2_launch_test.xml.xml: 1 tests
build/motoman_gazebo/test_results/motoman_gazebo/roslaunch-check_test_sia5_with_multi_kinectv2_launch_test.xml.xml: 1 tests
build/motoman_gripper/test_results/motoman_gripper/roslaunch-check_test_gripper_launch_test.xml.xml: 1 tests
build/motoman_mh5_support/test_results/motoman_mh5_support/roslaunch-check_test_launch_test.xml.xml: 1 tests
[... cut ...]
build/motoman_sia10f_support/test_results/motoman_sia10f_support/roslaunch-check_test_launch_test.xml.xml: 1 tests
build/motoman_sia20d_support/test_results/motoman_sia20d_support/roslaunch-check_test_launch_test.xml.xml: 1 tests
build/motoman_sia5d_support/test_results/motoman_sia5d_support/roslaunch-check_test_launch_test.xml.xml: 1 tests
Summary: 36 tests, 0 errors, 0 failures, 0 skipped
```
## まとめ
1. 解決したこと
- とりあえずローカルでビルドとテストを通す
-  cloneしてる`motoman`パッケージでもテストがコケる問題
2. 残り課題
- `industrial_ci`にsub folderを設定する方法を探す

RyodoTanaka, 2018-01-29:
上記のローカルでとおるようにした際に，`motoman_project`以外も修正しました．
なので，改めて
```bash
wstool merge -t src
```
などでパッケージを更新してください．でないと失敗します．

RyodoTanaka, 2018-01-29:
# 結局うごかない...orz
ローカルだと上記の方法でうまくゆくのですが，Travisだと下記のようにパスが見つかってないようなエラーが出ます．
https://travis-ci.org/Nishida-Lab/motoman_project/builds/334817654#L8285-L8294
なんでそうなるのかわかりません（泣）

RyodoTanaka, 2018-01-30:
@MoriKen254 
早速の検証ありがとうございます．
https://github.com/MoriKen254/industrial_ci/tree/0e69ddf3044fb84594380e8c576df55deff2627b
上記の変更とTravisのログを確認しました．
現時点での気付きを記しておきます．
### 1. `source /root/catkin/devel/setup.bash` について
@MoriKen254 さんのされているとおり，私も`ROS_PACKAGE_PATH`に関係する事かと最初推測しました．
しかし，下記のログでも確認できるように，
```bash
source /root/catkin_ws/devel/setup.bash
```
を行うと`ROS_PAVKAGE_PATH`が`/`だけになるようですね...
よくわかりません．

### 2. `ROS_PACKAGE_PATH`について
ローカルのうまく行っている環境で`ROS_PACKAGE_PATH`を調べました．
##### ただし，`source devel/setup.bash`をしていなくても，`catkin run_tests`時のパスは通っていました．
ので，あまり関係無いことになってしまうのですが，一応結果を記しておきます．
- `source devel/setup.bash`やる前
   ```bash
   ROS_PACKAGE_PATH=/opt/ros/indigo/share:/opt/ros/indigo/stacks
   ```
- `source devel/setup.bash`やった後
   ```bash
   ROS_PACKAGE_PATH=/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/dhand_ros_pkg/dhand_control:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/dhand_ros_pkg/dhand_description:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/dhand_ros_pkg/dhand_driver:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/dhand_ros_pkg/dhand_gazebo:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/dhand_ros_pkg/dhand_msgs:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/dhand_ros_pkg/dhand_ros_pkg:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_recognition/motoman_cable_removal:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_config:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_control:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_demo_msgs:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_demo:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_description:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_gazebo:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_gripper:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_mh5_support:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_moveit:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_msgs:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_driver:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_recognition/motoman_point_cloud:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_sda10f_support:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_sda10f_moveit_config:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_sia10d_support:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_sia10f_support:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_sia20d_support:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_sia20d_moveit_config:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_sia5_moveit_config:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_sia5_moveit_plugins:/home/ryodo/workspace/ROS/test_ws/src/motoman_tools/motoman/motoman_sia5d_support:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_viz_msgs:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_rviz_plugins:/home/ryodo/workspace/ROS/test_ws/src/motoman_project/motoman_viz:/opt/ros/indigo/share:/opt/ros/indigo/stacks
   ```

### 3. そもそも `source /root/catkin_ws/devel/setup.bash` の挙動がよくわからない
1.項にも記していますが，`source /root/catkin_ws/devel/setup.bash`をしたにもかかわらず`ROS_PACKAGE_PATH`が`/`だけになるという挙動がよくわかりません．
何より，`source /root/catkin_ws/devel/setup.bash`した直後は`/`だけなのに[あらためて`ROS_PACKAGE_PATH`を見てみる](https://github.com/MoriKen254/industrial_ci/blob/0e69ddf3044fb84594380e8c576df55deff2627b/industrial_ci/src/tests/source_tests.sh#L188-L199)と[元に戻っている](https://travis-ci.org/Nishida-Lab/motoman_project/jobs/334956163#L3098-L3103)という挙動はさらによくわかりません．

MoriKen254, 2018-01-30:
@RyodoTanaka 
ありがとうございます．

さすが，察しが良いですね．私の意図は正にその通りで，まとめてもらってありがたいです．更に，補足情報ももらえて助かります．
結果，一体この挙動をどう解釈すればよいのか，さっぱり分からないのです笑．そもそもDockerが分からないのですが汗．

launch testを通そうと思ったら，そろそろ付け焼き刃は限界で，本家にissueを上げる段階まで来ているかもしれませんね…．

RyodoTanaka, 2018-02-01:
@MoriKen254 
お返事遅くなってすみません．
> 結果，一体この挙動をどう解釈すればよいのか，さっぱり分からないのです笑．そもそもDockerが分からないのですが汗．
私もまったくこのとおりです...orz

> 本家にissueを上げる段階まで来ているかもしれませんね…．
本家に聞いてみるのに賛成です．
時間出来次第聞きます．
もし，お時間があれば @MoriKen254 さんから本家に聞いていただけるととても助かります．
よろしくお願いします．

130s, 2018-02-03:
口出ししておきながら間があいてしまいました汗

launch test と言われているのがなんの事を指すのかちょっと分かっていないので，関係あるかわかってない中で分かり難い点を補足だけしますと，industrial_ci だと devel ではなく install スペースを使います (catkin_make でいうと `install` オプション付きの状態)．で，`{catkin_make, catkin} run_tests` は install スペースに対応していないので，industrial_ci は独自に `.test` ファイルを探して `rostest` に渡すようになっています．参考になるかわかりませんが．．．

ユーザ側で BEFORE_SCRIPT 等でハックをなるべくしなくて済むようにできるだけ usecase を拾って対応していきたいので，issue 開いて下さるの大歓迎です :)

130s, 2018-02-03:
あと depend 漏れは http://wiki.ros.org/roslaunch/#Catkin がおすすめです．

RyodoTanaka, 2018-02-06:
@130s 
遅くなってすみません．
ご教授いただき，本当にありがとうございます！
まずはdepend漏れを確認した後，ダメならissueで質問します．

RyodoTanaka, 2018-03-06:
@MoriKen254 @130s 
修論などでかなり時間が空いてしまいました．
すみません．

industrial_ci リポジトリのissueに質問を投稿しました．
https://github.com/ros-industrial/industrial_ci/issues/273

質問の仕方がよろしくないかもしれないので，加筆や修正などでお助けいただけると嬉しいです．
よろしくお願いします．

RyodoTanaka, 2018-05-23:
@MoriKen254 @130s 
かなり時間が空いてしまい申し訳ありません．
解決方法を見つけました．
簡単に説明すると，`CMakeLists.txt`にInstallに関する項目を記述する必要があるということです．
以前upしていた`industrial-ci`の本家リポジトリのIssueでも報告していますので，良ければご覧ください．
https://github.com/ros-industrial/industrial_ci/issues/273

本当は本パッケージで行うべきだったのですが，規模が大きかったので，[CIR-KIT/fourth_robot_pkg](http://github.com/CIR-KIT/fourth_robot_pkg/tree/indigo-devel)で検証を行ってました．
結果，`fourth_robot_pkg`では，launchテストを追加してもきちんと動くようになりました．
本パッケージでの実装はもう少し先になるとは思いますが，とりあえず解決したと思います．
よろしくお願いします．

MoriKen254, 2018-07-06:
@RyodoTanaka さん

遅くなって申し訳ございません。GitHubのページを見たのも久しく汗。
ご依頼の通り、私から本家に聞けばよかったですね。修論等で忙しい時期に、お力になれなくて申し訳ない。

launchテストの通し方について、見えてきましたね。

本質は、Installですね。分かってしまえばああそうかと言うことですが、ローカルだとcatkin buildでうまく行ったものだから、Travisだと明示的にInstallしないといけないという発想が出てきませんでした汗。

すごく勉強になりました！粘り強く取り組んでくださり、ありがとうございます ^^

RyodoTanaka, 2018-07-06:
@MoriKen254 
お久しぶりです！
いえいえ，私が言い始めたことなので気になさらないでください！

- `motoman_project`だと規模が大きすぎる
- そもそも分割して管理したかった

という理由から，下記のOrganizationでmotoman_projectの主要プログラムを管理し始めました．
https://github.com/ntl-ros-pkg
現在置いてあるものであれば，すべてTravisで動作します！
ある程度完成したら，勉強も兼ねてrosdistroに載せれるようにしたいなと思ってます！

MoriKen254, 2018-07-06:
@RyodoTanaka 

ntl-ros-pkg、いいですね！
最近はこれとか、ROS勉強会を頑張っていたのですね！

これを参考に、cirkit_unit03側のlaunch testにも応用させてもらいます。

rosdistroに入れる方法が分かったら、教えてくださいね。

RyodoTanaka, 2018-07-06:
@MoriKen254 
ありがとうございます！
実は最近は
- つくばチャレンジ（個人参加）
- WRS トンネル災害チャレンジ
- OUXT Polaris （海上のつくちゃみたいなやつ）

を研究以外にやっています！
ちょっと多いですが，なんとか回して頑張ってます．
OUXT Polarisの方で，rosdistroにパッケージをリリースする手順を最近メンバーの方がされたりしてるので，割と早めにやり方をお伝えできるかと思います．

とりあえず私達が参考にしているページTORKのページです．
http://opensource-robotics.tokyo.jp/?p=3272

RyodoTanaka, 2018-07-06:
ROS勉強会はタダで参加したい！
ってのも発表の動機だったりします笑

MoriKen254, 2018-08-05:
追伸

以下のエラーの問題が残っていましたね。

```bash
Errors     << motoman_euclidean_cluster:make /root/catkin_ws/logs/motoman_euclidean_cluster/build.make.000.log
In file included from /root/catkin_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/euclidean_cluster.cpp:2:0:
/root/catkin_ws/src/motoman_project/motoman_recognition/motoman_euclidean_cluster/src/../include/euclidean_cluster.hpp:19:55: fatal error: pcl/features/moment_of_inertia_estimation.h: No such file or directory
 #include <pcl/features/moment_of_inertia_estimation.h>
```

一回無理やりインクルードファイル引っ張ってきても、今度はリンクで失敗してしまいました。

CMakeLists.txt で以下のラインが抜けていたからっぽいです。

```
include_directories(
  include
  ${catkin_INCLUDE_DIRS}
  ${OpenCV_INCLUDE_DIRS}
  ${PCL_INCLUDE_DIRS}　←　コレ
)

link_directories(${PCL_LIBRARY_DIRS})　←　念の為

target_link_libraries(euclidean_cluster_node
  ${catkin_LIBRARIES}
  ${OpenCV_LIBRARIES}
  ${PCL_LIBRARIES}　←　コレ
)
```

この設定が抜けている状態でなぜローカルでビルドが通っていたのか、そっちのほうが不思議です…。

落ち着いたらこのIssueも閉じましょう。

MoriKen254, 2019-07-06:
これもよさ気ですね。クローズ！

closed 2019-07-06 by MoriKen254