Nishida-Lab/motoman_project #114: hardware_interface::EffortJointInterface Issue

MuratCA77, 2018-03-15:
When I run the sia5_sim_control.launch file with sia5_joint_position_control.yaml, it is giving me the following error:

It is failed to load sia5_controller and could not find the resource “joint_s” in ‘hardware_interface::EffortJointInterface’.

I am using Ubuntu Linux 16.04 and Gazebo 7. Can you please help me fix this issue?

RyodoTanaka, 2018-03-15:
@MuratCA77 
# The reason of the Error
## Error1
> It is failed to load sia5_controller

That error message, says we could not  use `sia5_controller`, when we want to use `sia5_joint_position_control.yaml` configuration file.
So, we need to change the controller names on launch file.

## Error 2
Now, we use `position_controllers/JointPositionController` type for gazebo robot model that you can check [here](https://github.com/Nishida-Lab/motoman_project/blob/indigo-devel/motoman_description/urdf/sia5/body.transmission.xacro#L10).
But, on `kinect-devel` branch, `sia5_joint_position_control.yaml` file sets `effect_controllers/JointPositionController` mistakenly.
So, we need to fix from `effect_controllers/JointPositionController` to `position_controllers/JointPositionController`. 

# How to fix them ?
## Fixing Error1
Rewrite your `<motoman_project>/motoman_control/launch/sia5_sim_control.launch` file as following.
```xml
<launch>

  <!-- Load joint controller configurations from YAML file to parameter server -->
  <rosparam file="$(find motoman_control)/config/sia5/sia5_joint_position_control.yaml" command="load"/>

  <!-- load the controllers -->
  <node name="sia5_controller_spawner" pkg="controller_manager"
		type="spawner" output="screen" 
		args="joint_state_controller
	                  joint_s_position_controller
                          joint_l_position_controller
                          joint_e_position_controller
                          joint_u_position_controller
                          joint_r_position_controller
                          joint_b_position_controller
                          joint_t_position_controller"/>

  <!-- convert joint states to TF transforms for rviz, etc -->
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"
		respawn="false" output="screen">
  </node>

</launch>

```

## Fixing Error 2
Now, I patched the error on 3a7a0650f63ef7bacecabe95106af800122fd9ed.
So, pull it and launch it again.

# At the End
Now, we have a problem on testing our packages....
That means, there should be more errors.
So, if you find any other errors or bugs, please be free to open the Issue !

MuratCA77, 2018-03-15:
Hello Ryodo,

It was able to load the controllers but here is the problem. How can I control the speed of the robot? For example, when I change the PID gains in the yaml file, it does not even show its effects.

<launch>

RyodoTanaka, 2018-03-17:
@MuratCA77 
> How can I control the speed of the robot?

Now, we set the controller to be a Position Controller.
That means, we control position, not velocity (speed).
So, If you want to control velocity, you need to change controller that is written in `<motoman_description>/urdf/sia5/body.transmission.xacro`.

> when I change the PID gains in the yaml file, it does not even show its effects.

That sounds strange, in usual, if you set the pid gain, it shows the effects.
How does it reacts, if you set every gain to be 0 ?

MuratCA77, 2018-03-19:
Hello Ryodo,

I have changed the controller to velocity in the body.transmission.xacro file and controller .yaml file. I can set PID gains in RQT GUI which seems to be working properly. When it is a position controller, there is no option for PID gains based on Gazebo tutorials.

On the other hand, when I change the controller to effort controller in body.transmission.xacro file and .yaml file, the robot arm is falling on the ground all of a sudden.

Could you please also tell me how to implement a gripper for pick and place kitting application?

Thanks,
Murat