Nishida-Lab/motoman_project #118: 本リポジトリがROS教育用サービスで利用されているようです

MoriKen254, 2018-07-31:
嬉しいニュースです。

たまたま見つけたのですが、下記のROS教育サービスで、ここのmotomanがチュートリアルとして利用されているようです。

[image]

https://www.youtube.com/watch?v=_ULJrmJ7xOo&feature=youtu.be

コマンドラインやlaunchファイル名も確認しましたが、見覚えのあるモノばかり。先端もジャミングだし、間違いなく西田研のものを使っています。

完成度が認められているということで、誇っていいことだと思います ^^ 世界中の人が見ているかもしれませんね。

コレを糧に、もっと進化させたいですね！

このIssueは、少し様子を見た後に閉じます。以上、情報共有でした。

TakeshiNishida, 2018-07-31:
ROSが5分で使えるという内容ですね！また、べた褒めですね。「素晴らしい！詳しくどうやってるかわかんないけど」というのにはすこし笑いました。

Ry0, 2018-08-01:
無駄にリアルなロボット台も世界デビューできてうれしそうです。頑張ってアセンブリしてよかった！

MoriKen254, 2018-08-02:
:smile:

MoriKen254, 2018-08-05:
レスありがとうございました :laughing: 

閉じます。

closed 2018-08-05 by MoriKen254