Nishida-Lab/motoman_project #119: motomanをV-REPで動かしてみようのコーナー！

MoriKen254, 2018-07-31:
タイトル通りです。疲れたときに、趣味的にぼちぼちやれるネタがあればと思い、テーマアップです ^^

↓とりあえずWikiに今できていることをメモ。
https://github.com/Nishida-Lab/motoman_project/wiki/motoman%E3%82%92V-REP%E3%81%A7%E5%8B%95%E3%81%8B%E3%81%99%E3%81%BE%E3%81%A7%E3%81%AE%E8%BB%8C%E8%B7%A1

先日購入頂いたMasteringのROS本の解説、実は間違っているので注意してください笑。上記Wikiの通りにやれば、最低限ROSとのインターフェースはうまくいくはずです。

「え、Gazeboで既に動いているのに、V-REPでやる意味あるの？」

わかりません。でも、やるんです笑！楽しそうです。

体感的にですが、V-REPは動作が軽いように思います。デフォルトでロボットやワークの色々なモデルも用意されていますし、複数台ロボット構成もGazeboよりやりやすそうです。

将来的に製造ラインをバーチャルで再現することを考慮すると、実はV-REPのほうが引っ張ってこれる既存リソースは豊富そうですね。

↓こんなの再現できたら、めちゃ熱そう？
[image]

↓Youtubeも！ベルトコンベアが動いている！
[image]
https://www.youtube.com/watch?v=1qXLBXyxJ2E

Gazebo慣れしている身だと少し癖はありそうですが、一度乗っかってしまえばどのプロジェクトでも使えるようになるので、研究室としての恩恵は大な予感です。Gazeboと比較して、何ができて何ができないのかを把握できるようになると嬉しいなぁと思います。

というわけで、ちょっと存在感が薄れてきつつある（←オイ）motomanくんの方でも、たまにはただ楽しそうだということもやる感じで ^^ 趣味路線なので、本業のバックグラウンドで急がず焦らずまったりと。でもやるからにはクソまじめに。

TakeshiNishida, 2018-07-31:
V-REPを利用している研究機関は多いです。特に、人を登場させたい場合には、必須になります。Gazeboではできませんので。自律移動ロボットも、こちらの方がいいかもしれません。

MoriKen254, 2018-08-02:
ありがとうございます！徐々に移行していきましょう！