Nishida-Lab/motoman_project #123: ブランチ整理について

TakakiNishio, 2018-08-20:
ブランチ多すぎ問題を何とかしたい。
**個人でfork→適宜プルリク** での運用に切り替えるか？についても..
[こちらの議論 ](https://github.com/Nishida-Lab/motoman_project/pull/120#issuecomment-413521573)からのissueです。

TakakiNishio, 2018-08-20:
`demo/stable` は `indigo-devel` のビルドが通るまでの安定版ブランチでした。それをずるずると使い続けている状況で、実機の運用ブランチは現在も`demo/stable` です。`indigo-devel` にマージせねば..
 ~~さっさと kinetic に乗り換えろよっていう話ですよね..~~

ブランチ数の爆発を防ぐための **個人でfork→プルリク** での運用については賛成です。

MoriKen254, 2018-08-20:
ブランチの作成者は、消して良いブランチを自己申告しましょう。

## indigo 
- [fix-travis-indigo](https://github.com/Nishida-Lab/motoman_project/tree/fix-travis-indigo)
- [add-gitlabci-indigo](https://github.com/Nishida-Lab/motoman_project/tree/add-gitlabci-indigo)
- [mk/fix-ci](https://github.com/Nishida-Lab/motoman_project/tree/mk/fix-ci)
- [mk/add_travis_kinetic](https://github.com/Nishida-Lab/motoman_project/tree/mk/add_travis_kinetic)
- [mk/fix-ci-test](https://github.com/Nishida-Lab/motoman_project/tree/mk/fix-ci-test)

kinetic 周りはマージされるまで念の為残しておきたく…。

## kinetic への移行
kineticへ移行期間については、なんかあったときのために既存のindigoは据え置きにしたほうが良いと思います。何か起きて動かなくなったら大変そうなので。

TakakiNishio, 2018-08-20:
### 消して良いブランチ
- [takaki/kinetic-stomp-chomp](https://github.com/Nishida-Lab/motoman_project/tree/takaki/kinetic-stomp-chomp)
- [takaki/new_sia5](https://github.com/Nishida-Lab/motoman_project/tree/takaki/new_sia5)

### kinetic への移行
現在indigoで動いているデモの環境は残します。

AtsukiYokota, 2018-08-20:
## 消して良いブランチ
- [yokota/devel](https://github.com/Nishida-Lab/motoman_project/tree/yokota/devel)

MoriKen254, 2018-08-21:
少し様子を見ていた消して良いブランチの宣言については、8/21(水) 17:00 を締切とし、それ以降削除ということでいいですかね。

削除自体は、 @TakakiNishio さんお願いします。

次の段階として「消さないでほしいブランチ」の宣言に移行してもいいかな、と思います。これも期限を区切って、宣言されなかったものはバッサリ削除する方向で。

TakakiNishio, 2018-08-21:
承知しました。
消して良いブランチは 8/21(水) の 17:00 以降に順次削除します。

### 消してさないでほしいブランチ
- [demo/stable](https://github.com/Nishida-Lab/motoman_project/tree/demo/stable)
現在のデモが動く唯一のブランチのため

8月中にはバッサリ削除します。

MoriKen254, 2018-09-06:
一旦消していいブランチを共有。

## kinetic-devel のマージを記念して
- [add-gitlabci-to-kinetic](https://github.com/Nishida-Lab/motoman_project/tree/add-gitlabci-to-kinetic)
- [fix-kinetic-devel-2](https://github.com/Nishida-Lab/motoman_project/tree/fix-kinetic-devel-2)
- [fix-kinetic-devel](https://github.com/Nishida-Lab/motoman_project/tree/fix-kinetic-devel)
- [gitlab-ci/kinetic-devel](https://github.com/Nishida-Lab/motoman_project/tree/gitlab-ci/kinetic-devel)

## indigo-devel のマージを記念して
- [demo/stable](https://github.com/Nishida-Lab/motoman_project/tree/demo/stable)

@TakakiNishio くん
で、合ってそうかな？

TakakiNishio, 2018-09-07:
@MoriKen254 さん

ありがとうございます！

挙げて頂いたブランチは全て削除できますので、
今晩中にでも削除しておきます。

MoriKen254, 2018-09-09:
ありがとうございます。超絶きれいになりましたね！

閉じましょう。

closed 2018-09-09 by MoriKen254