Nishida-Lab/motoman_project #126: Gazebo上でD-Handでpick & placeしたい（オブジェクト付き）

MoriKen254, 2018-12-30:
すみません、標記を実現できるサンプルはありませんか？できれば対象のオブジェクトもあったりすると嬉しいのですが。

moveit でプランニングしたあと、実際に把持させるとき、具体的にどういう司令をd-hand側に送っているのかを把握したいなと思っていまして。

あるいは、超簡単なグリッパを先っちょにつけたモデルなんかがあったら嬉しいなと思うのですが、多分ないですよね。その時は作ってみようと思います。

Ry0, 2018-12-31:
https://github.com/qboticslabs/mastering_ros
学生時にこのあたりの参考書で把持を含む動作のサンプルがあったような気がします．

また同じようなことをIBMハッカソンで実現しようとして
同じプロセスを踏んだブランチがあったような．．．
もう削除しましたかね
りょどがDhandは受動関節があるから無理で，そのあと平行グリッパをつけたみたいな流れが

MoriKen254, 2018-12-31:
ありがとうございます。助かります。

そうですね、そのサンプルがシンプルで良さげですね。

そうそう、そのときにやったブランチがあればなと思ったのですが。。

現状のリポジトリにないようでしたら、参考に作りますね。

RyodoTanaka, 2019-01-08:
@MoriKen254 
すみません．
今気づきました．

DhandでGazebo上の物体をつかむ件ですが，下記理由により，正攻法ではうまく行かないです．
- Dhandの手先のCollisionモデル形状に平坦な面が無いので，掴みづらい．正確には，掴もうとしてもモデルが崩壊したりしてうまくつかめない．
- @Ry0 が言及してくれている通り，Dhandはそもそも受動関節&閉リンク機構が入っているので，実機と同じような司令（直動サーボの司令）だけで動くようにするには，何かしらの追加コードが必要．
- そもそも論で申し訳ないのですが， Dhandのモデル自体，実測しながら半分目分量でモデリングしているので，実機とのモデル誤差がとても大きい．故に，シミュレーションできたとしても，実機との差が大きそう．

で，解決策ですが，下記が考えられます．
1. いわゆる吸着プラグインを利用して掴んだことにする
2. DhandのCollisionモデルとは別に，並行グリッパに相当する，「把持専用の仮想joint」を設け，これを使って把持する．

現実的には，1. の解が現実的かと思います．
もし，Pick & Placeの速度を高め，そのシミュレーションが重要ということでしたら，2. の方法が良いかと思います．（拘束なPick & Placeで把持物体が落下することも考えられるため）

で，1. の吸着プラグインですが，一昔前までは[非公式の個人リポジトリ](https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin)のものが使われていたようですが，
最近，個人的に吸着グリッパをGazebo上に登場させて遊びたくなって調べてたら，下記パッケージ内にも実装されているようです．
http://wiki.ros.org/osrf_gear
ただ，いらない機能もあるので，吸着プラグインの部分だけ頂いて，自分でコーディングするのが最適解かなと思って最近取り組んでます．

今回の問題ですと，前者の
https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin
のプラグインを使うのが一番手っ取り早くて良いと思います．
以上，蛇足含んでますが，よろしくお願いします．

MoriKen254, 2019-03-17:
レス遅くなり申し訳ございません。

アドバイスありがとうございます。（このissue、大晦日に投下していたみたいですね笑。）

その後紆余曲折を経て、最終的には助言頂いた Jennifer さんの吸着プラグインを使うことにしました。Gazeboで検証するレベルならこれで十分だと判断しました。

実際、吸着プラグインなしのGazeboのデフォルト設定では、把持判定が厳しすぎて、チューニング地獄で心が折れました笑。

さて、当該吸着プラグイン使っているのは、https://github.com/ntl-ros-pkg 側の、下記箇所です。

- robot 側からの呼び出し
  - https://github.com/ntl-ros-pkg/motoman_robot/blob/master/motoman_description/robots/motomini/motomini_with_gripper_kinectv2.urdf.xacro#L14
- urdf 側での設定
  - https://github.com/ntl-ros-pkg/motoman_robot/blob/master/motoman_description/urdf/tool/gzplugin_grasp_fix.urdf.xacro

というわけで、本 Issue は閉じます。ありがとうございました。

closed 2019-03-17 by MoriKen254